lowRISC/ibex #601: Optimize CSRs and performance counters for FPGA Implementation

vogelpi, 2020-02-11:
The CSRs and most importantly performance counters map poorly on FPGA. We should investigate the effect of the performance counter width on resource utilization and timing (64-bit counters are known to map badly to FPGA) and see if they can e.g. be implemented differently (Xilinx DSP slices can be used to implement 48-bit counters).

ganoam, 2020-02-19:
Hi everyone,

The current core design features two mandatory 64 bit counters (mtime and minstret) and up to 29 optional counters with parameterizable width. It has been observed, that the resource utilization of the design is drastically increased by implementing a growing number of counters.

I have been investigating possibilities to optimize the implementation. One way of reducing the utilization is to implement the counters using DSP slices. The DSP48E1, featured by our example Artix 7 series device is capable of implementing counters of up to 47-bit width. However making synthesis tools infer those slices requires a few changes to the code.

* DSP slices do not implement counters by default. In order to signal the desire to use DSP slices, the `(* use_dsp = "yes"*) ` pragma is added to the counter register signal `mhpmcounter_q`.

* Counter increment and load logic and flop elaboration code must reside in the same generate block in order for vivado to recognize the possibility of using the DSP slice.

The benefits of using DSP slices for the optional counters of varying width are listed in the following tables. `Baseline` refers to the baseline design, `Modified DSP` refers to the design modified with the above changes and setting the `use_dsp` pragma and  `Modified no-DSP` refers to the modified design without the `use_dsp` pragma. In all reported designs, the maximum number of 29 optional counters was instantitated for maximum effect. The two mandatory 64-bit counters are always implemented in logic fabric. In order to show the exact effect of the modifications on the `cs_registers` module, the synthesis has been carried out using the `FLATTEN_HIERARCHY none` flag. Using the default (`rebuilt`) sightly lower total utilization can be achieved. However, savings are observed analogously. Implementation was done using the vivado `performance_explore` strategy.

The following table presents results for 29 32-bit wide optional counters.
```
              | Baseline | Modified no-DSP | Modified DSP
--------------+----------+-----------------+-------------
Total LUTs    | 3919     | 4062            | 3298
LUTS (cs_reg) | 1647     | 1789            | 1029
Total FFs     | 2939     | 2839            | 1911
FFs (cs_reg)  | 1343     | 1343            | 415
DSPs (cs_reg) | 0        | 0               | 29

```

The following table presents results for 29 48-bit wide optional counters.
```
              | Baseline | Modified no-DSP | Modified DSP
--------------+----------+-----------------+-------------
Total LUTs    | 4665     | 4637            | 3650
LUTS (cs_reg) | 2392     | 2367            | 1381
Total FFs     | 3303     | 3303            | 3303
FFs (cs_reg)  | 1807     | 1807            | 1807
DSPs (cs_reg) | 0        | 0               | 29

```

The following table presents results for 29 64-bit wide optional counters.
```
              | Baseline | Modified no-DSP | Modified DSP
--------------+----------+-----------------+-------------
Total LUTs    | 5251     | 5337            | 5369
LUTS (cs_reg) | 2982     | 3065            | 3098
Total FFs     | 3767     | 3767            | 3767
FFs (cs_reg)  | 2271     | 2271            | 2271
DSPs (cs_reg) | 0        | 0               | 58

```

For your reference: The resource utilization of the core with just the two mandatory 64 bit counters. (Also using the flag `FLATTEN_HIERARCHY none`).

```
              | Baseline |
--------------+----------+
Total LUTs    | 2963     |
LUTS (cs_reg) | 690      |
Total FFs     | 1882     |
FFs (cs_reg)  | 386      |
DSPs (cs_reg) | 0        |

```

Observations:

* Implementing 29 32-bit counters in logic fabric cost almost a third more resources than the baseline design. Implementing them 64 bit wide cost 77% more logic resources.

* Implemented in logic fabric, the an additional flipflop is used for every counter bit.

* For widths up to 48-bit, setting the `use_dsp` pragma drastically reduces logic utilization at the cost of one DSP slice per counter.

* For 32-bit counters, the FFs can be incorporated into the DSP primitive, resulting in better utilization of both logic LUTs and FFs. For 48-bit, surprisingly, this is not the case: the output pipeline register is deactivated in favor of fabric-implemented flip-flops. I have investigated the issue and have come to the conclusion that this is most probably due to the manner in which the netlist for loading new values is elaborated:
    - Loading a new value into the counter is done separately for bits `31:0` and `47:0`.
    - Rather than using a global clock-enable signal for all bits and feeding back the register output to the input for the bits that are not updated, The elaborated netlist describes this funcionality by masking out the corresponding clock-enable signals. Since the DSP output pipeline register only features a global clock-enable, Synthesis fails to infer usage of the output pipeline register for widths greater than 32 bits.
  I have tried to describe the counter in many different ways trying to get the correct behaviour and have not yet succeeded. If anybody knows how to do this, I am interested, but I think this is not worth any more time. - But still, the savings in terms of Logic LUTs is considerable.

* For 64-bit counters (actually everything above 48-bit), the utilization of logic LUTs is very similar to the baseline implementation. 48 bit happens to also be the limit of the DSP implementation of Xilinx's BinaryCounter macro. (http://www.xilinx. [... cut ...] on.

You might observe that for non-DSP implementations there is a slight increase of logic utilization. I am confident, that I could finetune the code such that this difference disappears. However it is long past time to report some preliminary results now.

You can have a look at my proposed implementation on my github fork, direct link:
https://github.com/ganoam/ibex/blob/fpga-opt-perf-mon/rtl/ibex_cs_registers_proposition.sv

It would of course have to be rebased as well.

What do you think?

tomroberts-lowrisc, 2020-02-19:
Some good findings!

I would vote for adding these details to the documentation, but not adding any pragmas into the code.

It would be interesting to know what code changes are absolutely necessary. If you could re-shuffle the generate blocks but leave the counter code as-is then I think that would be fine. Otherwise I think the next_value logic in your branch is a bit less readable (as well as increasing utilization) compared to the current implementation.

ganoam, 2020-02-21:
I have spent some more time investigating the situation. I have almost gotten rid of the increased utilization for non-DSP implementations and I may have identified the condition for vivado to infer DSP slices.

I have observed, that in every version which successfully infers DSP slices has one of the following structure:

```
assign load_sig = ...
assign updated_counter_sig = ...

always_comb begin
  if (load_condition) begin
    counter_d = load_sig
  end else if (update_condition) begin
    counter_d = updated_counter_sig;
  end else begin
    counter_d = old_counter_sig;
  end
end

always @(posedge clk_i) begin
  if (!rst_ni) begin
    counter_q <= '0;
    ...
```

or directly in the synchronous block:
```
assign load_sig = ...
assign updated_counter_sig = ...

always @(posedge clk_i) begin
  if (!rst_ni) begin
    counter_q <= '0;
  end else if (load_condition) begin
    counter_q <= load_sig
  end else if (update_condition) begin
    counter_q <= updated_counter_sig;
  end else begin
    counter_q <= old_counter_sig;
  end
end

```
The relevant points, as far as I understand:
- The If -Else structure must be present. Assigning a default value and conditionally reassigning the signal does not work.
- The assigned values must be prepared in advance. Assigning `counter_d = counter + 64'h1;` directly in the if-else structure does not work.

Also, note that the DSP output pipeline register reset synchronously.

The `Modified` design used to generate the reported results is given the following file: http://github.com/ganoam/ibex/blob/fpga-opt-perf-mon/rtl/ibex_cs_registers_proposition-rebased.sv

There is some flexibility on how to write the inference logic. However, of all my trials, this gave the most favourable utilization results.

I have synthesized and implemented the design again with- and without the DSP pragma. I have based the core on the FPGA register file and the FF register file. The numbers are reported for both configurations in the format `FPGA / FF`.

The signal `counter_sig` may be a next-value signal, to be assigned in a synchronous `always @` block, or the

The following table presents results for 29 32-bit wide optional counters.
```
              | Baseline       | Modified no-DSP | Modified DSP
--------------+----------------+-----------------+-------------
Total LUTs    | 3410 / 3959    | 3391 / 4095     | 2873 / 3340
LUTS (cs_reg) | 1603 / 1652    | 1585 / 1794     | 1058 / 1036
Total FFs     | 1847 / 2839    | 1847 / 2839     | 919 / 1911
FFs (cs_reg)  | 1343 / 1343    | 1343 / 1343     | 415 / 415
DSPs (cs_reg) | 0 / 0          | 0 / 0           | 29 / 29
crit. path /ns| 15.110/ 16.305 | 14.944/ 15.317  | 15.065/ 15.522
```

The following table presents results for 29 48-bit wide optional counters.
```
              | Baseline        | Modified no-DSP | Modified DSP
--------------+-----------------+-----------------+-------------
Total LUTs    | 3959 / 4707     | 4159 / 4692     | 3228 / 3692
LUTS (cs_reg) | 2161 / 2399     | 2354 / 2368     | 1426 / 1385
Total FFs     | 2311 / 3303     | 2311 / 3303     | 2311 / 3303
FFs (cs_reg)  | 1807 / 1807     | 1807 / 1807     | 1807 / 1807
DSPs (cs_reg) | 0 / 0           | 0 / 0           | 29 / 29
crit. path    | 15.774 / 15.359 | 15.588/ 15.288  | 15.961/ 15.881

```

The following table presents results for 29 64-bit wide optional counters.
```
              | Baseline        | Modified no-DSP | Modified DSP
--------------+-----------------+-----------------+-------------
Total LUTs    | 5129 / 5294     | 4682 / 5380     | 5042 / 5424
LUTS (cs_reg) | 3320 / 2987     | 2879 / 3069     | 3230 / 3102
Total FFs     | 2775 / 3767     | 2775 / 3767     | 2775 / 3767
FFs (cs_reg)  | 2271 / 2271     | 2271 / 2271     | 2271 / 2271
DSPs (cs_reg) | 0 / 0           | 0 / 0           | 58 / 58
crit. path    | 15.324/ 15.535  | 15.431/ 15.132  | 14.876 / 15.251

```

For your reference: The resource utilization of the core with just the two mandatory 64 bit counters. (Also using the flag `FLATTEN_HIERARCHY none`).

```
              | Baseline      |
--------------+---------------+
Total LUTs    | 2489 / 3001   |
LUTS (cs_reg) | 685  / 698    |
Total FFs     | 890  / 1882   |
FFs (cs_reg)  | 386  / 386    |
DSPs (cs_reg) | 0 / 0         |
crit. path    | 15.203/ 15.394|

```

Note:

* For the FPGA core configuration with non-DSP Counter implementations:
  - The Logic Lut utilization for 32 bit counters is almost exactly the same for baseline and modified implementations.
  - For 48 bit counters, the `Modified` implementation has a higher Logic LUT implementation. (+ 200)
  - For 64 bit however, utilization of the modified no-DSP implementation is lower(-447 Logic LUTS)
  To be honest I cannot explain the discrepancy between 48- and 64 bit implementations. I suspect there is some synthesis optimization that works or does not work depending on a power-of-two bit width, or similar. If you have an idea, I am interested.

* For the FPGA core configration with DSP Counter implementations:
  - We observe a reduction by 518 (537 compared to baseline) Logic LUTs and 928 FFs for 32-bit counters and by 931 (731 compared to baseline) and 0 FFs for 48-bit counters
  - The 64-bit implementation does not offer signicicant benefits.

* For the FF core configuration, the benefits of DSP Implementations are similar. The discrepancies between baseline and modified non-DSP impementations are less pronounced, but still observable.

Let me know if there is anything else you want to see to decide if or how you want to adapt the modifications.

GregAC, 2020-02-21:
Thanks for your work on this @ganoam 

I think it's also worth exploring options that will break strict RISC-V spec compatibility but will be perfectly reasonable for some applications, e.g. making mcycle and minstret have a configurable width so they can be less than 64-bit (or allow removing them altogether, potentially separately, I can see a good case for a 32-bit or less mcycle and no minsret for example).

It would also be interesting to look at what else in the cs_registers could be shrunk down, again allow breakage with the spec.

I'd really like to fit an Ibex into an ICE40 and I suspect these kinds of optimisations will be required to make that happen.

vogelpi, 2020-02-25:
These are very interesting findings and the overall savings in terms of LUTs and FFs are considerable (18% and 46%, respectively for 29 32-bit counters). I would like to get these code changes integrated, especially since the actual RTL changes are minimal.

We might need to rerun the Yosys flow to make sure it can still optimize the unused counter flops etc, i.e., we should make sure that these changes won't undo PR #541. Maybe @GregAC could do that once @ganoam has prepared a PR?

We should describe in the documentation, how to get the most efficient implementation on FPGA (32-bit counters only) and mention to optionally reduce even the counter width for mcycle and minstret which will break compatibility, as suggested by @GregAC .

We should also add a new parameter, e.g. called FPGA_XILINX or maybe we can use a VIVADO_SYNTHESIS ifdef, to actually insert the required DSP pragma. But this is a detail, and we can reason about it when @ganoam has prepared the PR.

ganoam, 2020-02-25:
Ok, I have prepared a PR #624. I have implemented it using a preprocessor variable instead of a regular parameter, since synthesis appears to have problems with signal declarations within generate-if blocks. I have added the parameter to the FPGA example's `.core` file. The default value is irrelevant, however it seems to be necessary to assign something in order for the variable to be set. (https://www.gitmemory.com/issue/olofk/fusesoc/263/491425454) I am not particularily happy with this solution, but I have so far failed to have a better idea.

Also, I have resynthesised the design setting the directive `FLATTEN_HIERARCHY rebuilt`, as is done in a standard synthesis run. This results in slightly different utilization results which I have quoted in the documentation. Since utilization figures per module are no longer accurate this way, the savings I quoted are to be understood with respect to the total core size.

One more thing: Bits `[11:MHPMCounterNum]` of the signal `mhpmcounter_incr` are not defined (for `MHPMCounterNum > 8`). Left like this, synthesis will not result in the utilization figures I reported, since those counters are optimized away. I order to generate my results, I have added the following lines to force synthesis of 'dummy counters'.
```
for (int unsigned i=11; i<3+MHPMCounterNum; i++) begin
  mhpmcounter_incr[i] = 1'b1;
end
```

vogelpi, 2020-03-06:
PR #624 has finally been merged. This issue can be closed. Thanks @ganoam for the great work!

closed 2020-03-06 by vogelpi