lowRISC/ibex #621: FPGA example for RISCV-compliant debug

jrrk2, 2020-02-21:
You announced that one of the first things that was done to ibex was make the debug RISCV debug compliant. However the debug block does not seem to be mentioned in the default fpga build, nor could I find a reference to an openocd config file or recommended version of riscv-dbg to be used with the recommended FPGA board. Did I get stuck in a backwater somehow ?

asb, 2020-02-21:
See [OpenTitan](https://github.com/lowRISC/opentitan) for a system that integrates Ibex with a RISC-V debug module.

jrrk2, 2020-02-21:
Thanks for the pointer Alex,
Your blog of [here](https://www.lowrisc.org/blog/2019/10/ibex-on-fpga-get-stuff-executed/) seems to point towards pulpissimo as the preferred reference platform. However that relies on pulp-builder which has an element pulp-runtime which is not publicly accessible.

asb, 2020-02-21:
That blog post predates the public announcement of OpenTitan, so were limited in what we could say! @vogelpi has been doing some work recently with Ibex and PULPissimo so is best placed to comment on the status there.

imphil, 2020-02-21:
To answer your specific questions: the latest version of riscv-dbg works just fine, the OpenOCD config file is in the opentitan repository, together with all things necessary to run that on a Nexys Video FPGA board, or simulate in a Verilator environment.

jrrk, 2020-02-21:
I have raised an issue on the pulp-platform side.

imphil, 2020-02-21:
Looks like there's nothing to be done on the Ibex side then.

closed 2020-02-21 by imphil