lowRISC/ibex #625: Optimize Bit-Shift Operations for FPGA Implementation

ganoam, 2020-02-25:
Hi everyone,

I have investigated the possibility to optimize the bit-shift operation on FPGA targets by using Multipliers to incorporate the logic. I have drafted two versions of this. 'NoFlatten' and 'Rebuilt' refer to the `FLATTEN_HIERARCHY` synthesis attribute used when generating the results.

* [ibex_alu_dspshift.sv](https://github.com/ganoam/ibex/blob/fpga-opt-dspbarrelshift/rtl/ibex_alu_dspshift.sv)
  Two 16x16 multiplier units describe left shifts. The resulting upper and lower bit arrays are reordered using the shift amount MSB. Concatenation is handled via addition, using one more DSP.
  ```
  DSP Utilization: 3
  Total LUT Utilization: 2478 / 2334 (NoFlatten / Rebuilt)
  ```

* [ibex_alu_circulardspshift.sv](https://github.com/ganoam/ibex/blob/fpga-opt-dspbarrelshift/rtl/ibex_alu_circulardspshift.sv)
 Cyclic barrel shifter, using 4 8x16 bit multipliers. The two MSB of `shift_amt` are used for the byte ordering MUX. Right shifts are done by shifting by `32-shift_amt`. The results are generated using a bitmask for shifted-out bits. Idea from: [Implementing Barrel Shifters Using Multipliers](https://www.xilinx.com/support/documentation/application_notes/xapp195.pdf)
 
 ```
 DSP Utilization: 4
 LUT Utilization: 2440 / 2339 (NoFlatten / Rebuilt)
 ```
The baseline implementation's resource utilization is:
```
Total LUT Utilization: 2486 / 2339
```

We can see a slight improvement when we prohibit hierarchy flattening during synthesis. However by when synthesising with the default settings, no improvements are observed. 

I suggest to not persue the issue further.

vogelpi, 2020-03-03:
Thanks @ganoam for reporting your findings here. I also think it is not worth to explore further in this direction.

vogelpi, 2020-07-15:
I am closing this issue. The conclusion here is to not investigate further.

closed 2020-07-15 by vogelpi