lowRISC/ibex #644: Try out svlint

imphil, 2020-03-02:
I just found https://github.com/dalance/svlint and would be interested how it copes with the ibex code base.

lukamac, 2020-10-27:
Hi!
I've tried running `svlint` with all rules (about 25 of them) enabled on all SystemVerilog files and these are the unique rules they fail:
- [case_default](https://github.com/dalance/svlint/blob/master/RULES.md#case_default)
- [enum_with_type](https://github.com/dalance/svlint/blob/master/RULES.md#enum_with_type)
- [function_with_automatic](https://github.com/dalance/svlint/blob/master/RULES.md#function_with_automatic)
- [generate_keyword_forbidden](https://github.com/dalance/svlint/blob/master/RULES.md#generate_keyword_forbidden)
- [generate_keyword_required](https://github.com/dalance/svlint/blob/master/RULES.md#generate_keyword_required)
- [genvar_declaration_out_loop](https://github.com/dalance/svlint/blob/master/RULES.md#genvar_declaration_out_loop)
- [inout_with_tri](https://github.com/dalance/svlint/blob/master/RULES.md#inout_with_tri)
- [input_with_var](https://github.com/dalance/svlint/blob/master/RULES.md#input_with_var)
- [legacy_always](https://github.com/dalance/svlint/blob/master/RULES.md#legacy_always)
- [level_sensitive_always](https://github.com/dalance/svlint/blob/master/RULES.md#level_sensitive_always)
- [loop_variable_declaration](https://github.com/dalance/svlint/blob/master/RULES.md#loop_variable_declaration)
- [output_with_var](https://github.com/dalance/svlint/blob/master/RULES.md#output_with_var)
- [parameter_in_package](https://github.com/dalance/svlint/blob/master/RULES.md#parameter_in_package)
- [unique_keyword](https://github.com/dalance/svlint/blob/master/RULES.md#unique_keyword)
- [wire_reg](https://github.com/dalance/svlint/blob/master/RULES.md#wire_reg)
I guess not all rules are important for this project, so maybe a few could be ignored.

imphil, 2020-10-27:
@lukamac we do have a svlint configuration in OpenTitan at https://github.com/lowRISC/opentitan/blob/master/.svlint.toml, which should use the same coding style as this repository. Can you try this configuration to see how it works?

lukamac, 2020-10-27:
This is what I get with the OpenTitan's .svlint.toml configuration:
- case_default
- enum_with_type
- function_with_automatic
- generate_keyword_forbidden
- level_sensitive_always
- loop_variable_declaration
- wire_reg
Altogether, there are 16 errors.

imphil, 2020-10-27:
Would you mind posting the whole list of errors/warnings? There might be some things which are worth fixing.

lukamac, 2020-10-27:
Wasn't sure how to post it so I've put all the failures in [fail.txt](https://github.com/lowRISC/ibex/files/5447902/fail.txt).
Hope that is OK :smiley:

lukamac, 2020-10-27:
I've put the output into [fail.txt](https://github.com/lowRISC/ibex/files/5447959/fail.txt).
Hope it's OK :smiley:

imphil, 2020-10-27:
Of course it is, thanks a lot! The warnings are actually valid, and should be fixed. (Well, except for the ones in the vendor directory, those need to be fixed in OpenTitan.) If you feel like it, I'd appreciate a pull request for those changes. Otherwise, I'll have a look in the coming days.

And maybe we can also add the svlint config to the ibex repository?

lukamac, 2020-10-27:
I'll fix them tomorrow and make a PR :smile: