lowRISC/ibex #647: Larger Prefetch buffer

Sarafim, 2020-03-03:
Hi,
I am just wondering if it is possible to use [prefetch buffer from riscy](https://github.com/openhwgroup/cv32e40p/blob/1f013cae7f49aca8e59ce32941924dc6b0aaec2a/rtl/riscv_if_stage.sv#L205) instead of [default one from ibex](https://github.com/lowRISC/ibex/blob/98c8cd39ea10e95fdecc8b1cea5eff1ca619a6a2/rtl/ibex_if_stage.sv#L137) ?

vogelpi, 2020-03-03:
Hi @Sarafim ,

I think it's worth trying out. The core-internal interface is very similar. The two main differences are:
1. Ibex does not support HW loops (you can probably just tie the `hwloop_*_i` signals to 0)
2. The 128-bit wide prefetch buffer does not support instruction fetch/PMP errors. I guess because the instruction cache it's usually connected to in PULP system does not either. You would have to feed constant 0 back to error input of Ibex.

Two more things to be aware of:
1. In Ibex we changed the behaviour of the I-side interface/prefetch buffer a while back to keep the address stable once the request signal is asserted. As far as I know, this has not yet been done for RI5CY and especially not for PULP systems. Your interconnect must be able to deal with that if you change the prefetch buffer. Please refer to #296 for more details.
1. The 128-bit wide prefetch buffer has a constant 1 input on the `req_i` input. I think in Ibex this can go low to disable instruction fetch when running into a WFI instruction and turning of the clock. I would expect that the 128-bit wide prefetch buffer is not able to properly handle that and that you might need a workaround. 

Other than that, I don't see an obvious problem. You might still run into some corner case issues. Good luck!

P.S.: In case you give it a try, it would be interesting to learn if it works, or what your findings are.

Sarafim, 2020-03-04:
What are the reasons, that you have decided to design 128bit prefetch buffer instead of 64 for riscy?  Are you planning to add larger prefetch buffer for ibex?

GregAC, 2020-03-04:
By larger do you mean wider or deeper?

You can already set the depth of the Ibex prefetch buffer, take a look at ibex_fetch_fifo. There is a parameter `NUM_REQS` that sets the number of outstanding iside requests Ibex can have set to 2 by default, but you could set this higher if desired. 

https://github.com/lowRISC/ibex/blob/cb9413e38bc43117a0a5dfd7221bbf8155815478/rtl/ibex_fetch_fifo.sv#L15-L17

The depth of the prefetch buffer is set by the number of requests but you can override this by altering the `DEPTH` parameter:

https://github.com/lowRISC/ibex/blob/cb9413e38bc43117a0a5dfd7221bbf8155815478/rtl/ibex_fetch_fifo.sv#L39

Is there a particular reasons you want to increase the size of the prefetch buffer? 

I'm currently experimenting with static branch prediction and this may require a wider prefetch buffer to aid timing closure but I haven't yet worked out the details.

vogelpi, 2020-03-04:
> What are the reasons, that you have decided to design 128bit prefetch buffer instead of 64 for riscy?

I think that question would be best asked to the PULP team. If I remember correctly, PULP systems use shared instruction caches with a cache line size of 128-bit. The prefetch buffer is basically an L0 cache and the i guess the width of 128-bit is a trade-off between size, pressure on the shared cache as well as implementation and verification complexity.

> Are you planning to add larger prefetch buffer for ibex?

I think there are no plans to add a wider prefetch buffer or instruction interface, but @tomroberts-lowrisc is in the process of designing a private instruction cache as a drop-in replacement of the prefetch buffer (actually, this work is almost done). Here is the PR: #585

Sarafim, 2020-03-04:
> By larger do you mean wider or deeper?

@GregAC  , by larger I actually mean wider.
In my project I have slow ROM, but I can configure its data width. It makes possible to read 2,3 or 4 instruction per 1 read cycle and will improve performance of the core. 

> I think there are no plans to add a wider prefetch buffer or instruction interface, but @tomroberts-lowrisc is in the process of designing a private instruction cache as a drop-in replacement of the prefetch buffer (actually, this work is almost done). Here is the PR: #585 

@vogelpi  , what do you mean by the **private**instruction cache? Are you going to replace prefetch  buffer in master branch?

vogelpi, 2020-03-05:
> @vogelpi , what do you mean by the **private** instruction cache? Are you going to replace prefetch buffer in master branch?

I think the plan is that it can be enabled using a parameter. But the default will probably keep using the current prefetch buffer.

Sarafim, 2020-03-06:
Is it possible somehow to create feature request for wider prefetch_buffer?

My thoughts are that it can be very useful for embedded applications, where cache usage is not acceptable because of the area constraints, but frequency are higher than memory response time.

imphil, 2020-03-06:
> Is it possible somehow to create feature request for wider prefetch_buffer?

Sure, you can do so by opening an issue here on the issue tracker. Please describe how you'd like the feature to look in as much detail as possible.

This feature can then be picked up by anyone looking for work. However, as long as we at lowRISC don't need this feature for our own work, we are unlikely to implement it in a timely manner.

So if you need it, you probably either need to implement it yourself (and open a pull request once you've done so!), or contract someone else do the implementation for you. If you'd like lowRISC to do it, please get in touch at info@lowrisc.org.

vogelpi, 2020-07-15:
There has been no activity here for more than 4 months now. I am thus closing the issue. Feel free to reopen it.

closed 2020-07-15 by vogelpi