lowRISC/ibex #649: Path from data_error_i to instr_req_o

GregAC, 2020-03-04:
As reported by this OpenTitan issue: https://github.com/lowRISC/opentitan/issues/1657 there may be a path from data_error_i to instr_req_o. We probably don't want this (for one thing it is a long feedthrough path that will cause timing issues).

tomroberts-lowrisc, 2020-03-04:
I would guess it is this code in the controller:
```
          if (special_req) begin
            // Halt IF but don't flush ID. This leaves a valid instruction in
            // ID so controller can determine appropriate action in the
            // FLUSH state.
            ctrl_fsm_ns = FLUSH;
            halt_if     = 1'b1;
          // set PC in IF stage to branch or jump target
          end else if (branch_set_i || jump_set_i) begin
            pc_set_o       = 1'b1;
```
`special_req` contains the bus error signals, and `pc_set` goes to the branch signal

GregAC, 2020-03-04:
Ah yes, makes sense. We don't attempt to jump to the exception handler til the cycle following the error but it's getting factored in anyway due to the else. I'll ponder how to fix it.

commit b932a8bb (2020-03-05): [rtl] Break path from data_err_o -> instr_req_o

Fixes #649
closed 2020-03-05 by GregAC