lowRISC/ibex #652: gdb support

jrrk2, 2020-03-04:
By default the ibex repository does not include riscv-dbg (gdb) support. Such support is available within the opentitan repository as part of a much larger project. I have modified the basic ibex example to include gdb support for a total of 4085 LUTs, by crudely copying relevant modules from opentitan and modifying to suit the original example style. I believe this contribution would be attractive to potential commercial customers who will naturally assume debug support as a prerequisite. The question is, how to format these changes (made in the build hierarchy) to enable merging with your no debug example (tested on Arty-A7-A100)

[my fork](https://github.com/jrrk2/ibex/tree/gdb)

imphil, 2020-03-09:
Thanks for your work on this @jrrk2. 

I'm not convinced the Ibex repository is the right place to put such an example. Adding debug support needs much more SoC infrastructure than we have currently, including an extended "bus" implementation. I'd therefore leave the simple system as simple as it is for now, and redirect people to more fully-featured SoC implementations if they are looking for examples on how to integrate a RISC-V Debug Implementation with Ibex, with OpenTitan and (probably) PULPissimo being two examples for that.

Maybe it would be a viable path forward to link these sample integrations in the docs somewhere? And maybe we can also include a more minimal Ibex example that you put together in #654 that you keep in a personal repository on GitHub?

jrrk2, 2020-03-10:
Closing this issue because the solution is discussed in https://github.com/lowRISC/ibex/pull/654

closed 2020-03-10 by jrrk2