lowRISC/ibex #669: Questa complains about newlines in ASSERT macros

imphil, 2020-03-09:
(Spin-off from #622 by @danghai)

Compiling Ibex with Questa gives the following warning:

```
** Warning: ** while parsing macro expansion: 'ASSERT' starting at /home/danghai/hhoangda/ibex/rtl/ibex_compressed_decoder.sv(286)
** at /home/danghai/hhoangda/ibex/rtl/ibex_compressed_decoder.sv(286): (vlog-2643) Unterminated string literal continues onto next line.
```

The code in question is

```verilog
// in ibex_compressed_decoder.sv
  `ASSERT(IbexInstrLSBsKnown, valid_i |->
      !$isunknown(instr_i[1:0]))

// in prim_assert.sv
`define ASSERT(__name, __prop, __clk = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST) \
`ifdef INC_ASSERT                                                                        \
  __name: assert property (@(posedge __clk) disable iff (__rst !== '0) (__prop))         \
    else `ASSERT_RPT(`PRIM_STRINGIFY(__name), `PRIM_STRINGIFY(__prop))                   \
`endif

// Converts an arbitrary block of code into a Verilog string
`define PRIM_STRINGIFY(__x) `"__x`"

`define ASSERT_RPT(__name, __msg)                                         \
`ifdef UVM                                                                \
  assert_rpt_pkg::assert_rpt($sformatf("[%m] %s: %s (%s:%0d)",            \
                             __name, __msg, `__FILE__, `__LINE__));       \
`else                                                                     \
  $error("[ASSERT FAILED] [%m] %s: %s", __name, __msg);                   \
`endif

```

Given this code, I'm assuming the error message actually comes from the fact that `prop` (the second argument to `ASSERT`) is converted to a string before being used in `ASSERT_RPT`, and this use of `prop` with newlines isn't liked by Questa.

According to SV2017, the definition of a string literal is 

[image]

Which indicates our code is actually wrong. I see a couple of options:
- Remove the reporting of the property in `ASSERT_RPT`, and only report the message.
- Add backslashes to the end-of-line of each multi-line property. I'm not sure if that's actually valid for the non-string case?
- Avoid newlines in properties when writing `ASSERT` macros.

@danghai:
- First, let's check if the hypothesis holds: Does the warning go away if you remove the `ASSERT_RPT()` call from the definition of the `ASSERT` macro?
- Can you try to add a backslash at the end of the property in `ibex_compressed_decoder.sv`, e.g. 

   ```systemverilog
     `ASSERT(IbexInstrLSBsKnown, valid_i |-> \
         !$isunknown(instr_i[1:0]))
   ```

Questa version info: QuestaSim-64 qrun 2019.4 Utility 2019.10 Oct 15 2019

CC @msfschaffner @sriyerg

sriyerg, 2020-03-09:
Looks like its between having the convenience of having the whole failing property showing up in the logs v/s striving for a cleaner code that is supported on all the tools. 

I personally like the convenience of being able to look at the failing property at a quick glance. It shows up in our reports as well:
https://reports.opentitan.org/hw/ip/usbdev/dv/latest/results.html

Although, the error message does indicate the file name and line number, so its not a lot of work to quickly lookup the failure by opening the file. 

Here's my preference for dealing with this:
1. Make all properties single line. I think 80-90% of the cases would probably fit within the 100 char line limit. For the rest, we use a backslash (I am trying this with VCS & Xcelium). 
2. Remove the stringified property from the error message. Less preferred, but I am ok with this.

rswarbrick, 2020-03-11:
Thinking about this, there's one other option I can see: use an "ifdef Questa" to remove the property from the error message, but just for Questa builds.

imphil, 2020-03-11:
Well, it's invalid SV syntax, so I'd rather not hope that tools other than Questa silently continue to accept it. 

@danghai please let us know the answers to the two questions above if you have time!

danghai, 2020-03-11:
Hi @imphil 
1) I do not see `ASSERT_RPT()` in rtl source code. I only see `ASSERT`.
2) I got the error when I try to do:
```
  `ASSERT(IbexInstrLSBsKnown, valid_i |-> \
      !$isunknown(instr_i[1:0]))
```

The error is:
```
 ** Error: (vlog-13069) ** while parsing macro expansion: 'ASSERT' starting at /home/danghai/ibex/rtl/ibex_compressed_decoder.sv(286)
```

imphil, 2020-03-11:
Thanks for testing the backslashes! So they don't work (as expected).

For the other test, can you apply this patch, which removes the use of `ASSERT_RPT` from the `ASSERT` macro.

```diff
diff --git a/shared/rtl/prim_assert.sv b/shared/rtl/prim_assert.sv
index f569292..9b5668f 100644
--- a/shared/rtl/prim_assert.sv
+++ b/shared/rtl/prim_assert.sv
@@ -82,7 +82,6 @@
 `define ASSERT(__name, __prop, __clk = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST) \
 `ifdef INC_ASSERT                                                                        \
   __name: assert property (@(posedge __clk) disable iff (__rst !== '0) (__prop))         \
-    else `ASSERT_RPT(`PRIM_STRINGIFY(__name), `PRIM_STRINGIFY(__prop))                   \
 `endif
 // Note: Above we use (__rst !== '0) in the disable iff statements instead of
 // (__rst == '1).  This properly disables the assertion in cases when reset is X at
```

danghai, 2020-03-11:
I got a lot of errors: (sample)
```
** Error: (vlog-13069) ** while parsing file included at /home/danghai/ibex/dv/uvm/core_ibex/env/core_ibex_env_pkg.sv(10)
** at /home/danghai/ibex/dv/uvm/core_ibex/env/core_ibex_rvfi_if.sv(27): near "endinterface": syntax error, unexpected "SystemVerilog keyword 'endinterface'".
** Error: (vlog-13069) ** while parsing file included at /home/danghai/ibex/dv/uvm/core_ibex/env/core_ibex_env_pkg.sv(11)
** at /home/danghai/ibex/dv/uvm/core_ibex/env/core_ibex_csr_if.sv(12): near "endinterface": syntax error, unexpected "SystemVerilog keyword 'endinterface'".
-- Importing package ibex_mem_intf_agent_pkg
-- Importing package irq_agent_pkg
** Error: (vlog-13069) /home/danghai/ibex/dv/uvm/core_ibex/env/core_ibex_env_pkg.sv(23): near "endpackage": syntax error, unexpected "SystemVerilog keyword 'endpackage'".

```

imphil, 2020-03-11:
Let's try this instead:

```
diff --git a/shared/rtl/prim_assert.sv b/shared/rtl/prim_assert.sv
index f569292..92a0545 100644
--- a/shared/rtl/prim_assert.sv
+++ b/shared/rtl/prim_assert.sv
@@ -82,7 +82,7 @@
 `define ASSERT(__name, __prop, __clk = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST) \
 `ifdef INC_ASSERT                                                                        \
   __name: assert property (@(posedge __clk) disable iff (__rst !== '0) (__prop))         \
-    else `ASSERT_RPT(`PRIM_STRINGIFY(__name), `PRIM_STRINGIFY(__prop))                   \
+    else $error("assert failed");                                                        \
 `endif
 // Note: Above we use (__rst !== '0) in the disable iff statements instead of
 // (__rst == '1).  This properly disables the assertion in cases when reset is X at
```

danghai, 2020-03-11:
I see the warning `vlog-2643` is removed when I apply this patch. However it still has other warnings (it does not relate to this patch)

```
 18 ** Warning: /home/danghai/ibex/rtl/ibex_cs_registers.sv(730): (vlog-2583) [SVCHK] - Extra checking for conflicts with always_comb     and always_latch variables is done at vopt time.
 19 ** Warning: /home/danghai/ibex/rtl/ibex_cs_registers.sv(732): (vlog-2583) [SVCHK] - Extra checking for conflicts with always_comb     and always_latch variables is done at vopt time.
 20 ** Warning: /home/danghai/ibex/rtl/ibex_cs_registers.sv(739): (vlog-2583) [SVCHK] - Extra checking for conflicts with always_comb     and always_latch variables is done at vopt time.
 21 ** Warning: /home/danghai/ibex/rtl/ibex_cs_registers.sv(742): (vlog-2583) [SVCHK] - Extra checking for conflicts with always_comb     and always_latch variables is done at vopt time.
 22 ** Warning: /home/danghai/ibex/rtl/ibex_cs_registers.sv(745): (vlog-2583) [SVCHK] - Extra checking for conflicts with always_comb     and always_latch variables is done at vopt time.

```

imphil, 2020-03-12:
Thanks for your tests @danghai. So we are back to the options @sriyerg outlined in https://github.com/lowRISC/ibex/issues/669#issuecomment-596685769: Remove the property from the message, or restrict all properties in `ASSERT` calls to a single line. 

I looked through the uses of `ASSERT` in Ibex and in OpenTitan (we want to keep the two aligned), and I've seen a fair number of asserts which are significantly longer than a single line. That's especially true for the asserts we're using in FPV, which use the same mechanisms. (If you want to have a look yourself, run `git grep -C3 'ASSERT('` in opentitan.)

With that I don't see options other than removing the property from the message. I'll prepare a PR for that to see how it looks.

imphil, 2020-03-12:
@danghai I've filed #696 to look into the last remaining warning in https://github.com/lowRISC/ibex/issues/669#issuecomment-597823073 more closely.

PR #699 (2020-03-13): Remove property from assert message
Assert macros in prim_assert.sv report the asserted property as part of
the error message if the assertion fails. Doing that requires us to
stringify the property. If the property is multi-line in the source
code, this produces a multi-line string. According to the
SystemVerilog standard, a multi-line string requires a backslash (`\`)
at the end of a line. Adding this backslash through a macro is not
possible (at least to my knowledge), giving us no way to produce a
standards-compliant implementation of the `ASSERT` macros that report
the property as string. [... cut]
commit 1c82c201 (2020-03-13): Remove property from assert message

Assert macros in prim_assert.sv report the asserted property as part of
the error message if the assertion fails. Doing that requires us to
stringify the property. If the property is multi-line in the source
code, this produces a multi-line string. According to the
SystemVerilog standard, a multi-line string requires a backslash (`\`)
at the end of a line. Adding this backslash through a macro is not
possible (at least to my knowledge), giving us no way to produce a
standards-compliant implementation of the `ASSERT` macros that report
the property as string [... cut]
closed 2020-03-13 by imphil