lowRISC/ibex #671: [tracking] Questa issues

imphil, 2020-03-09:
Questa issues reported in #622 by @danghai 

- [ ] #669 (Questa complains about newlines in ASSERT macros)
- [ ] #670 (Questa: Defaulting port to var rather than wire)

danghai, 2020-03-11:
I try to run the full flow in `Makefile`. I got the result so far:
[regr.log](https://github.com/lowRISC/ibex/files/4313779/regr.log)
Compile and Optimize log:
[qrun_compile_optimize.log](https://github.com/lowRISC/ibex/files/4313790/qrun_compile_optimize.log)
1 sample Sim log (all are the same, 2 warning messages)
[sim.log](https://github.com/lowRISC/ibex/files/4313794/sim.log)

imphil, 2020-03-13:
The current master should have all outstanding issues resolved, can you double-check if the Questa runs are now warning-free @danghai?

danghai, 2020-03-13:
Hi @imphil . It is now warning-free. I believe that you can close the issue. Thank you

imphil, 2020-03-13:
Great, thanks!

closed 2020-03-13 by imphil