lowRISC/ibex #674: [dv] minor Makefile issues

udinator, 2020-03-09:
Hi @rswarbrick, this issue is just meant as a tracker of sorts for some minor issues I may encounter with the updated Makefile flow. Admittedly, many of these potential issues will just be my relative unfamiliarity with the new simulation output format or other minor questions, so this thread will just act as a place to consolidate all of these questions.

udinator, 2020-03-09:
Currently, I'm seeing something interesting happen where if I run, for example, `make SEED=123 ITERATIONS=1 TEST=riscv_rand_instr_test`, followed by `make SEED=123 ITERATIONS=1 TEST=riscv_illegal_instr_test`, ensuring that the second test is different from the first one, both tests will output into `out/seed-123/rtl_sim`, but the second test will overwrite the results of the first test, so that the only test whose outputs I can actually see is the second test, or in this case `riscv_illegal_instr_test`. 
Was this intended? And would be possible to add the ability to run various single test instances as per the previous example into the same `seed-####` output directory? I feel this would be a nice quality of life improvement, that the old flow afforded users.

udinator, 2020-03-09:
One other small thing is, if I run some test, say with `make SEED=123 ITERATIONS=1 TEST=riscv_rand_instr_test`, and then modify the corresponding yaml test entry in `riscv_dv_extension/testlist.yaml` and try rerunning the test with the same command to keep the seed the same, the flow will immediately exit out, since no changes would have been detected with the metadata files.

rswarbrick, 2020-03-10:
> Currently, I'm seeing something interesting happen where if I run, for example, `make SEED=123 ITERATIONS=1 TEST=riscv_rand_instr_test`, followed by `make SEED=123 ITERATIONS=1 [...]

Hmm. The problem is that the Makefile doesn't understand the relationship between the value of the `TEST` variable and the files that `run.py` and `sim.py` create. I think we could probably make this a little less annoying by getting rid of the clean-up that runs at the start of the `gen` and `rtl_sim.compile.stamp` targets. That way,

    make SEED=123 TEST=foo
    make SEED=123 TEST=bar

would leave both `foo` and `bar` results. Of course, the Makefile still wouldn't really know what's going on, so running

    make SEED=123 TEST=foo

again would cause it to regenerate everything for test `foo`.

The other problem with this idea is that we wouldn't necessarily throw away generated files if we changed simulation options. I'm not sure whether that's a problem, but I wrote the Makefile to be conservative and rebuild if it wasn't sure.

What do you think?

rswarbrick, 2020-03-10:
> One other small thing is, if I run some test, say with `make SEED=123 ITERATIONS=1 TEST=riscv_rand_instr_test`, and then modify the corresponding yaml test entry in `riscv_dv_extension/testlist.yaml` and try rerunning the test with the same command to keep the seed the same, the flow will immediately exit out, since no changes would have been detected with the metadata files.
> EDIT: this also applies to running any of the individual steps in the flow after modifying only the yaml testlist entry

Ah, that's easy. I've just pushed up PR #678, which should fix it.

udinator, 2020-03-11:
> Hmm. The problem is that the Makefile doesn't understand the relationship between the value of the `TEST` variable and the files that `run.py` and `sim.py` create. I think we could probably make [...]

As far as the relationship between the `TEST` variable and the metadata files, would there be some way of keeping track of the names of the tests that have been run in one of the metadata files, appending a new directory under `out/seed-123/rtl_sim` if the testname is not present in the metadata, and overwriting the corresponding directory if the testname is present in the relevant stamp file?

> 
> The other problem with this idea is that we wouldn't necessarily throw away generated files if we changed simulation options. I'm not sure whether that's a problem, but I wrote the Makefile to be conservative and rebuild if it wasn't sure.

As far as this point goes, I would be in favor of not throwing away generated files if all we change are simulation options, e.g. if we run the RTL simulation and then decide we want to change some simulation-time plusarg in the testlist and rerun the RTL sim, we should just run the simulation step and not recompile and so on. Is something like this what you were referring to?

udinator, 2020-03-11:
One issue I'm now seeing is that I'm unable to generate waves or coverage data from my simulations,  I believe the issue lies in the `cov-arg, wave-arg, lsf-arg` variables that are defined on lines 313-315 of the Makefile as the `if` conditional expressions seem to always be evaluating to False, producing empty results.

rswarbrick, 2020-03-11:
I'll have a think about the regeneration question above and get back about that later in the day. For now, I'll have a look at the coverage/trace/LSF options and try to figure out what I miswired there.

rswarbrick, 2020-03-11:
That turned out to be easy - I'd just missed an 'S'. I think that the coverage arguments are getting passed correctly, though. Can you have another look: if it doesn't look right, could you give the command you're running? I've found that a good way to debug things like this is to run `make -n`: this means you see the commands that get run, and not the messages from the rather chatty scripts.

udinator, 2020-03-12:
> That turned out to be easy - I'd just missed an 'S'. I think that the coverage arguments are getting passed correctly, though. Can you have another look: if it doesn't look right, could you give the [...]

After your change to fix the typo, the --en_wave and --en_cov arguments are getting passed correctly into the sim.py script.
There is still an issue with the waveform generation, namely that the waveform information is getting dumped into the core_ibex/ directory, instead of into the out/seed-###/rtl_sim directory. My guess is some issue with the directory paths that are being passed into the vcs command, I'm looking into the problem.

rswarbrick, 2020-03-12:
@udinator I've moved discussion of avoiding some rebuilds to issue #689, so that this issue can stay focused on actual bugs (and can hopefully be closed one day!).

rswarbrick, 2020-03-12:
The remaining issue being tracked here (so far) is that dumped waves end up in `core_ibex` directory rather than somewhere inside `out`.

rswarbrick, 2020-03-12:
Pull #691 should fix the wave file paths.

udinator, 2020-03-13:
Another issue has popped up, this one a bit more more severe: because the Makefile chooses a single seed that it then assigns to every single RTL simulation and riscv-dv simulation, the generated coverage database for every simulation is then assigned the same name, as the database name is assigned based off of the seed, and needs to be unique for every test that is run. This then results in every test overwriting the previously generated coverage directory, rendering the final coverage directory completely inaccurate - applying to both RTL simulations and instruction generation simulations.
The coverage database name is given by `-cm_name test_<seed>` in yaml/rtl_simulation.yaml, which gives us a few options:
1) Assign a unique name for each `-cm_name` inside of the sim.py script itself that is independent of the seed. This would split coverage options between the YAML configuration file and inside of the sim.py script. This would also require modifications to the riscv-dv run scripting  to do the same thing, so I prefer that we avoid this solution.
2) Alternatively, we revert to the original mechanism of assigning each test a random seed inside sim.py itself instead of generating a top level seed inside of the Makefile which then propagates to every test, which gives us the required coverage database name uniquification.

@taoliug, what do you think about this issue?

rswarbrick, 2020-03-13:
Well, (2) works with probability 1-1/2^32! :-)

More seriously, it seems like you'd probably like to be able to generate coverage if you set the seed on the Makefile command line too, right?

Option (1) is easy to implement, and we don't need to split stuff between `sim.py` and `rtl_simulation.yaml`: can't we do the same trick as I just implemented for PR #691? I'm not quite sure how the coverage flow works: do you get a new coverage database for each run? In which case, I guess you'd want something like -cm_name `test_<seed>_<testname>_<iteration>` (at the moment, sim.py doesn't substitute `<testname>` or `<iteration>`, but that's trivial to add).

udinator, 2020-03-13:
Yep, it would still be nice to set the seed from the command line as well.
Essentially, a coverage database is generated for every test generated by the instruction generator, and for every RTL simulation, and which all need unique names.
Something I'm thinking as well is similar to your idea, tagging the coverage directories with <testname>_<iteration> with the trick you used in #691.

rswarbrick, 2020-03-16:
Am I right in thinking that the coverage question has been resolved by PR #704? (And, in which case, is there anything else left on this tracker?). If we see new bugs, maybe it would be better to open a new issue for each: trawling through a long list of comments is hard work and slightly defeats the point of using an issue tracker! :-)

udinator, 2020-03-16:
Yep, #704 should resolve the coverage issue. I will close this issue as well.

closed 2020-03-16 by udinator