lowRISC/ibex #679: Error in tests riscv_debug_basic_test

Evgenii666, 2020-03-10:
Hello, I am running your tests and have encountered a set of errors. The **riscv_debug_basic_test.0.o** test fails, giving the following message:

_UVM_FATAL /var/home/korsakov-ea/designs/lowrisc_centOS/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(196) @ 27330: uvm_test_top [uvm_test_top] Did not receive core_status INITIALIZED within 500 cycle timeout period_

I looked into the code of the **test** class and found that you really expect to receive **INITIALIZED** status from the core:

In **core_ibex_test_lib.sv** file is written:

  _virtual task send_stimulus();
    fork
      begin
        vseq.start(env.vseqr);
      end
      begin
        if (cfg.require_signature_addr) begin
          **wait_for_core_setup();**
        end else begin
          …_

  _**virtual task wait_for_core_setup();**
    wait_for_csr_write(CSR_MSTATUS, 10000);
    core_init_mstatus = signature_data;
    // capture the initial privilege mode ibex will boot into
    init_operating_mode = priv_lvl_e'(core_init_mstatus[12:11]);
    wait_for_csr_write(CSR_MIE, 5000);
    core_init_mie = signature_data;
    **check_next_core_status(INITIALIZED, "Core initialization handshake failure", 500);**
  endtask_

In **core_ibex_base_test.sv** file is written:

_virtual task **check_next_core_status**(core_status_t core_status, string error_msg = "",
                                      int timeout = 9999999);
    run.raise_objection(this);
    fork
      begin
        wait_for_mem_txn(cfg.signature_addr, CORE_STATUS);
        signature_data = signature_data_q.pop_front();
        `DV_CHECK_EQ_FATAL(signature_data, core_status, error_msg);
      end
      begin : wait_timeout
        clk_vif.wait_clks(timeout);
        `uvm_fatal(`gfn,
                   $sformatf("Did not receive core_status %0s within %0d cycle timeout period",
                   core_status.name(), timeout))
      end
    join_any
    // Will only get here if we successfully beat the timeout period
    disable wait_timeout;
    run.drop_objection(this);
  endtask_

From the last it is seen that you expect to receive **core_status** **INITIALIZED** within 500 clock cycles. However, if you look at the code for the assembler program, you will notice that such a transaction is sent by the processor only once at the end of the **init** function:

_init:             
                  li x0, 0x0
                  li x1, 0xf34ac426
                  li x2, 0x80000000
                  li x3, 0x9
                  li x4, 0x5cede9c9
                  li x5, 0xfbd6e9d7
                  li x6, 0xfe7970bc
                  li x7, 0xf78f116f
                  li x8, 0xd
                  li x9, 0x4
                  li x11, 0xd19db6b3
                  li x12, 0x0
                  li x13, 0x2
                  li x14, 0x0
                  li x15, 0x80000000
                  li x17, 0x0
                  li x18, 0x0
                  li x19, 0x80000000
                  li x20, 0xa71e4c5e
                  li x21, 0x80000000
                  li x22, 0xfe090abc
                  li x23, 0xa0eb3051
                  li x24, 0xd69a0cbd
                  li x25, 0xf93399d2
                  li x26, 0x6c85c9e9
                  li x27, 0x2
                  li x28, 0x80000000
                  li x29, 0x0
                  li x30, 0x80000000
                  li x31, 0xfec1a8bd
                  la x10, user_stack_end
                  j main
                  **li x24, 0x8ffffffc
                  li x31, 0x0
                  slli x31, x31, 8
                  addi x31, x31, 0x0
                  sw x31, 0(x24)**_

Firstly, the processor does not have enough time allotted to you even to get to the **j main** command (time runs out on the **li x31, 0xfec1a8bd** command). Secondly, as I understand it, the processor will never execute this piece of code, because after executing the **j main** command it will switch to the execution of the **main** function code.

Maybe I'm wrong. Could you explain why the test fails?

udinator, 2020-03-10:
Hi @Evgenii666, thanks for bringing this issue up. If you sync to the latest upstream branch, you will find that this issue has been fixed.

Evgenii666, 2020-03-11:
Thanks, I updated the repository and restarted the tests.

Over the past two weeks, I have updated your repository and related external projects about 5 times. During all this time, two times the tests were not compiled due to errors generated by the gcc compiler, three times the tests ran, but always with errors. The number of tests with errors varies from several tens to two hundred (partly the IUS 15.2 simulator that I use is to blame for this). After the last update, waves for tests ceased to be created for me.

Perhaps it would be much more convenient to work with your project if you only committed verified and working versions to the master branch, and made intermediate changes in other branches.

rswarbrick, 2020-03-11:
Hi Evgenii,

You are right that this test-suite is currently rather unstable, but you should find that the RTL itself is not. If you want to use Ibex to run your own binaries, you might find the `simple_system` flow (in `examples/simple_system` more useful. Eventually, the randomised tests in `dv/uvm/core_ibex` should stabilize, but that hasn't happened yet.

Evgenii666, 2020-03-11:
Well, thanks for the advice.

closed 2020-03-11 by Evgenii666