lowRISC/ibex #687: [tracking] Use Verible's verilog_format for code formatting

imphil, 2020-03-11:
Verible's verilog_format is a automatic code formatter. We want to be able to run it on our code base, ultimately in "enforcing" mode, requiring all submissions to be formatted with this tool. This is the same approach we take with clang-format and C/C++ files. This issue is to track the progress towards that.

Ibex TODO:
- [ ] Add Verible format support to edalize
- [ ] Add a fusesoc target to our core files
- [ ] Add Verible format in non-enforcing mode to CI
- [ ] Document how to use it
- [ ] Clean up existing code once the auto-formatted style matches our expectations.
- [ ] Switch to enforcing mode.

Blocking Verible bugs:
- [ ] https://github.com/google/verible/issues/228

imphil, 2020-07-03:
The first set of code changes to get our code in line with what verible format produces is at https://github.com/lowRISC/ibex/pull/1014. The goal is to have that PR cover all the "easy" cases, getting us step-by-step closer to a full equivalence.

asb, 2020-07-04:
Thanks for your work in pursuing this, filing upstream issues, and tracking progress.

IMHO #351 should be higher priority - it seems to be a case of producing incorrectly formatted code, which is a much bigger barrier to enforcing verible formatting use than producing output that doesn't match our personal tastes. But once incorrect formatting bugs are addressed I agree with the thrust of your argument in #1014 - don't sweat the small stuff, and in return formatting becomes a solved problem.

We'll want to consult with downstream consumers (OpenTitan) prior to landing a large cosmetic change like #1014. I don't see why it should be problematic at this stage, but it couldn't hurt to explain the plan.