lowRISC/ibex #698: Command for sanity regression in CI

imphil, 2020-03-12:
We'd like to run a sanity regression in CI, with a runtime of 5-15 min. 

@udinator what command should we use for this run?

udinator, 2020-03-12:
Hi @imphil , for a basic sanity regression that runs one iteration of every test in the testlist, the command is `make ITERATIONS=1`.

imphil, 2020-03-13:
`make -C dv/uvm/core_ibex ITERATIONS=1 ISS=spike SEED=1` gives me a runtime of ~ 20 min on my machine. Probably good enough for a start, but we should try to not increase that time much more.

imphil, 2020-08-20:
On CI, I get a runtime of ~42 minutes, which is rather on the long side. 

@udinator do you have ideas to cut this time down?

udinator, 2020-08-20:
Yep, one idea would be to not run a full sanity regression, but just run a few select tests from the testlist that exercise some major features of the design.
For example, we could have the CI command be 
```console
make -C dv/uvm/core_ibex ITERATIONS=1 TEST=riscv_rand_instr_test,riscv_multiple_interrupt_test,riscv_debug_basic_test
```

This would run three specific tests to just make sure that the normal execution pipeline works correctly, as well as basic interrupt and debug handling capabilities are still up and running.