lowRISC/ibex #700: Cannot get seed value by default from Makefile

danghai, 2020-03-13:
I got error when I run the Makefile with default seed value. It seems it does not generate random seed value. Reproduce:

```
danghai@ubuntu:~/ibex/dv/uvm/core_ibex$ make SIMULATOR=qrun VERBOSE=1 OUT=output
mkdir -p output/seed-/.metadata
usage: run.py [-h] [--target TARGET] [-o O] [-tl TESTLIST] [-tn TEST]
              [--seed SEED] [-i ITERATIONS] [-si SIMULATOR] [--iss ISS] [-v]
              [--co] [--cov] [--so] [--cmp_opts CMP_OPTS]
              [--sim_opts SIM_OPTS] [--gcc_opts GCC_OPTS] [-s STEPS]
              [--lsf_cmd LSF_CMD] [--isa ISA] [-m MABI]
              [--gen_timeout GEN_TIMEOUT]
              [--end_signature_addr END_SIGNATURE_ADDR] [--iss_opts ISS_OPTS]
              [--iss_timeout ISS_TIMEOUT] [--iss_yaml ISS_YAML]
              [--simulator_yaml SIMULATOR_YAML] [--csr_yaml CSR_YAML]
              [--seed_yaml SEED_YAML] [-ct CUSTOM_TARGET]
              [-cs CORE_SETTING_DIR] [-ext USER_EXTENSION_DIR]
              [--asm_tests ASM_TESTS] [--c_tests C_TESTS]
              [--log_suffix LOG_SUFFIX] [--exp] [-bz BATCH_SIZE]
              [--stop_on_first_error] [--noclean] [-d DEBUG]
run.py: error: argument --seed: invalid int value: ''
Makefile:222: recipe for target 'output/seed-/.metadata/instr_gen.gen.stamp' failed
make: *** [output/seed-/.metadata/instr_gen.gen.stamp] Error 2
```

rswarbrick, 2020-03-13:
Hmm, this is strange. I can't reproduce it here. Please can you give the SHA of your checkout and run make with the '--trace' argument (which gets it to print the commands it runs)?

dawidzim, 2020-03-13:
hey!
change in `Makefile` line number `15` to this:
`SEED                := $(shell bash -c 'echo $$RANDOM')`

this works for me

rswarbrick, 2020-03-13:
Ok, now I understand (Ubuntu machines link /bin/sh to dash, but my Debian box pointed at /bin/bash). I've just pushed up PR #701, which should fix things.

rswarbrick, 2020-03-13:
@danghai : This should be fixed now. Could you check? If so, I'll close this.

danghai, 2020-03-14:
Thanks. I believe it is resolved.

closed 2020-03-14 by danghai