lowRISC/ibex #703: [DV] Prerequisites for bitmanip DV

imphil, 2020-03-13:
We are making progress on implementing the bitmanip (B) extension in Ibex in #89. This issue tracks things we need to have before we can start DV work on it.

- [ ] A compiler toolchain with bitmanip support in its assembler (at least). Tracking in https://github.com/lowRISC/lowrisc-toolchains/issues/8
- [ ] Bitmanip support in spike and ovpsim (that exists AFAIK, we might need to adjust the build instructions)
- [ ] Bitmanip support in riscv-dv.

imphil, 2020-03-13:
@taoliug @udinator what are the plans for bitmanip in riscv-dv?

udinator, 2020-03-13:
In riscv-dv, we've had a PR (https://github.com/google/riscv-dv/pull/498) opened by an external contributor to add most of the support for B-extension instructions. @weicaiyang is going to help with merging this PR into the codebase soon, and then I will take over B-extension verification for Ibex from there.

imphil, 2020-03-23:
An updated compiler toolchain is available in CI after https://github.com/lowRISC/ibex/pull/724 is merged. If you want to manually get one, download a release 20200323-1 or newer from https://github.com/lowRISC/lowrisc-toolchains/releases.

vogelpi, 2020-04-23:
FYI: I have tested the Bitmanip support in Spike and RISCV-DV. The linked PR enables it and adds a tests for Ibex.

What is missing is support for functional coverage for the B instructions. But people are working on that upstream (RISCV-DV).

vogelpi, 2020-07-03:
Spike/OVPSim support for bitmanip can be enabled and we documented how to do this. RISCV-DV has support for bitmanip support.

I think we are done here, WDYT @imphil ?

closed 2020-07-03 by imphil