lowRISC/ibex #709: Wrong mtval recorded for instruction access fault

tomroberts-lowrisc, 2020-03-16:
The RISCV Privileged ISA specification specifies:
"For instruction-fetch access or page-fault exceptions on systems with variable-length instructions, `mtval` will contain the virtual address of the portion of the instruction that caused the fault while `mepc` will point to the beginning of the instruction."

In Ibex currently, both `mepc` and `mtval` will point to the beginning of the instruction, even if the fault originated in the second half of a misaligned uncompressed instruction.

I see two solutions:
1) Wire `mtval` to zero on instruction access faults to indicate that this reporting feature is not supported.
2) Capture the second address somehow in this particular case. (It will always be the ID stage PC + 2)

vogelpi, 2020-03-17:
Hi @tomroberts-lowrisc ,

If I understand this correctly, the problem can only occur for misaligned uncompressed instructions as these are split up into two aligned 32-bit instruction fetches, both of which can cause e.g. a PMP, memory, or ECC error.
 
I think the case that only the second instruction fetch causes an error is quite a special case but we probably still want to have that. It helps recovering and debugging, i.e., it makes working with the core easier.

I thus would go for Option 2. What would be your preference?

tomroberts-lowrisc, 2020-03-17:
Yeah I think you're right, it would be best to have this info.
I'll have a think if there's a way of doing it without adding a load of extra logic

GregAC, 2020-03-17:
I agree, best to follow the spec, doesn't look like there's an option not to provide it?

vogelpi, 2020-03-18:
> I agree, best to follow the spec, doesn't look like there's an option not to provide it?

I think it would be fine to set it to 0 and signal that we don't support it for that particular case as @tomroberts-lowrisc pointed out. The spec says (Sec. 3.1.17, p. 38): "The hardware platform will specify which exceptions must set mtval informatively and which may unconditionally set it to zero."

But I think it's valuable to provide this information to the programmer/fault handler here.

PR #713 (2020-03-18): [rtl] Fix mtval for unaligned instr errors
mtval should record which half of the instruction caused the error
rather than just recording the PC.
An extra signal is added in the IF stage to indicate when an error is
caused by the second half of an unaligned instruction. This signal is
then used to increment the PC by 2 for mtval capture on an error.

Fixes #709
commit 42aa761c (2020-03-18): [rtl] Fix mtval for unaligned instr errors

mtval should record which half of the instruction caused the error
rather than just recording the PC.
An extra signal is added in the IF stage to indicate when an error is
caused by the second half of an unaligned instruction. This signal is
then used to increment the PC by 2 for mtval capture on an error.

Fixes #709
closed 2020-03-18 by tomroberts-lowrisc