lowRISC/ibex #726: Makefile error when using different TESTLIST

udinator, 2020-03-25:
Hi Rupert,
If I run `make SEED=123 ITERATIONS=1 TESTLIST=riscv_dv_extension/ml_testlist.yaml TEST=riscv_rand_test COV=1 ISS=spike all`, it results in this error:
`make: *** [Makefile:379: ml_out/seed-123/regr.log] Error 1`.
The test itself does not pass, which is a separate issue, but I'm more interested in seeing what is happening in the Makefile, specifically the `$(OUT-SEED)/regr.log:` target that throws a make error, as this error short circuits the rest of the Makefile and causes coverage generation to be skipped.
I am pretty sure it has something to do with the fact that a different testlist is being specified here, but I'm not quite sure how this is resulting in a make error.

rswarbrick, 2020-03-25:
@udinator I'm afraid I can't reproduce this. I've attached what I see with head of master (`dbfe8de`). If there's not something silly going on, please could you reproduce the error and attach a terminal log?

[issue-726-log.txt](https://github.com/lowRISC/ibex/files/4379996/issue-726-log.txt)

udinator, 2020-03-25:
@rswarbrick you actually are reproducing the error, if you look in the terminal log you attached, right above your call of "git rev-parse" at the very bottom of the file, you'll see the make error that I mentioned.

rswarbrick, 2020-03-25:
Oh, I see. Well that's not a "make error": it's Make telling you that the command that's supposed to create `regr.log` exited with a nonzero status code, which makes sense: the tests failed. You'll see that the regression log exists (with a failing test).

Which part of this behaviour is not as you'd expect?

udinator, 2020-03-25:
Ah that's a silly mistake on my end then, I guess I got confused between make telling me that the command failed and thinking that something in the makefile itself failed. I'll figure out a workaround and close this issue then.

closed 2020-03-25 by udinator