lowRISC/ibex #730: [DV] UVM environment should obey/understand mtvec alignment constraints

GregAC, 2020-03-25:
I am seeing a failure in the riscv_illegal_instr_test. The root of the issue is Ibex jumps to a different PC upon seeing an illegal instruction compare to spike. There are further tests with similar looking failures that I suspect have the same root cause.

The reason for this is in the particular test I'm looking at mtvec is set to 0x80001229 at the beginning of the test.

The privileged spec allows implementation defined alignment of mtvec base

> An implementation may have different alignment constraints for different modes. In particular,
MODE=Vectored may have stricter alignment constraints than MODE=Direct.

On Ibex mtvec is hardcoded into vectored mode (mtvec[1:0] == 2'b01) and mtvec base must be aligned to a 256-byte boundary so the [7:0] bits are ignored when writing an mtvec value.

This results in Ibex jumping to 0x80001200 and spike (which seems to accept any 4-byte aligned address for base) jumping to 0x80001228 leading to a mismatch.

A simple fix would be for the UVM environment to always generate an appropriately aligned mtvec but this would mean no testing of the scenario where mtvec is written with a mis-aligned base. It would be preferable if it could understand instead that when a mis-aligned base is written to mtvec the exception handler will be located at the aligned base address instead. This could give some difficulties it matching up traces with spike however.

udinator, 2020-03-25:
Hi @GregAC , I think this is a result of my changes to the instruction generator to enable PMP support inadvertently destroying the 256-byte boundary of the exception  handler and instead causing traps into the middle of the NMI interrupt handler. I'm currently working on a fix for this.

udinator, 2020-03-26:
Update: This issue has been fixed with #733 , the test should be okay now