lowRISC/ibex #745: Add a CI configuration for the B extension

imphil, 2020-03-27:
With work on the B extension landing we should add a CI configuration that tests this config. It can use the new framework added by @GregAC in #714.

imphil, 2020-03-27:
Maybe we can have a config with all of the experimental features.

GregAC, 2020-03-27:
Could even write a few basic self-checking tests for it? A good excuse to add a new piece of CI which runs a given set of binaries and reports pass/fail based on some success criteria?

imphil, 2020-03-27:
There are some tests in the bitmanip repository, including tests that look like they could work with the riscv-compliance framework at https://github.com/riscv/riscv-bitmanip/tree/master/compliance/rv32b. Maybe we can try dumping this folder into the compliance test suite and give it a try?

imphil, 2020-03-27:
But a first step is to at least synthesize code with RV32B enabled to ensure that we don't regress in lint or produce code that doesn't even synthesize any more :-)

vogelpi, 2020-03-30:
I think @ganoam has already a couple of such self-checking tests as he started the implementation before RISCV-DV had support for Bitmanip.

ganoam, 2020-03-30:
Yes, I do have a set of tests, but they are not really self-checking.

What I do is I synthesize a 'simple system' and let it run a set of bitmanip instructions. I write out the operands and the result to a log file, and check the result by reading the log and running the same commands on the host pc.

You can have a look at it [here](https://github.com/ganoam/ibex/tree/bitmanip-zbt-devel/examples/sw/simple_system/bitmanip_test). I did not antizipate it being used by anyone but me, so it is not very cleanly organized, and it features paths that are specific to my machine. It uses the common.mk file used for the simple system example in a slightly modified form. (most probably breaking the simple system example, I have not checked.)

To run tests:
* adapt the paths to your B-extension compiler ../common/common.mk 
* synthesize the simple system example
* comment or uncomment the instructions you want in test_zbb.c and test_zbt.c (leaving all of them in may take a long time)
* run `make distclean` and `make verify`.

The test result will be written to `[...]/bitmanip_test/out/report.txt`

commit 230e8796 (2020-03-31): [ci] Add Ibex with RV32B to configurations

Note this doesn't introduce any testing of the RV32B instructions,
simply runs existing tests on a configuration with the RV32B extension
enabled.

Fixes #745
closed 2020-03-31 by GregAC