lowRISC/ibex #753: Deal with the replications of Ibex parameter sets across .core files

GregAC, 2020-03-31:
We have multiple .core files that each have the same set of Ibex parameters (RV32E/RV32M/RV32B/MultiplierImplementation etc).

Ideally these would all get defined in one place that can then be reused. Uncertain if fusesoc has existing functionality to deal with this or is this something to request/add ourselves?

imphil, 2020-03-31:
There is no fusesoc functionality for that at this point. The problem is similar to https://github.com/olofk/fusesoc/issues/275 and could probably be solved with a "include a yaml fragment" feature. I don't think that's part of standard YAML, but I haven't looked too closely at this point. Another piece of related work could be the [templating mechanism of Azure Pipelines](https://docs.microsoft.com/en-us/azure/devops/pipelines/process/templates?view=azure-devops).

olofk, 2020-12-09:
I can think of two ways to accomplish this but I'm not sure they will fit your purposes.

1. Add a core which sets all these parameters and have other cores depend on it
2. Invoke a generator from the top-level core that sets these from something, e.g. from parameters in the core file or from an external file