lowRISC/ibex #763: What's wrong with the IRQ ISR return?

zzyhotmail, 2020-04-05:
It seems that the program can NOT be continued correctly after the interrupt return, where the interrupt is happened during the normal function return.

My project is from Opentitan system on chip. In my program, the timer interrupt is used.

The following ASM code  is a normal function calling and return progress. This function was called by main. I dump it from the core tracing. The first line “PC: 20014ce” is the function’s entry. From line 14 “PC: 20001584” to line 23 “PC: 20001596” are the function return progress.

[https://github.com/zzyhotmail/file/blob/master/normal_func.PNG](url)

The problem is that: during the previously function return process (after line 14), a timer IRQ happened. The ISR was called and executed. As shown in the following code. The ISR function entry is line 15 “PC: 2000011c”.

[https://github.com/zzyhotmail/file/blob/master/isr_entry.PNG](url)

As my understanding, after the ISR return, the program should be continue from line 15, in the first figure, just follows line 14 where the IRQ happened.
But, the fact is: after ISR return, several instructions in the function return progress are missed (Line 15 ~ Line 23). As shown in the following code (code_3), the program directly return to the main functions. 
[https://github.com/zzyhotmail/file/blob/master/isr_ret.PNG](url)

So, what’s wrong?

rswarbrick, 2020-04-06:
Hi there! It's going to be quite hard for us to help you figure out what's going on from just this information. We think that our interrupt handling works correctly (and even have some tests for this), so it would be rather difficult for us to dig down further without more help.

Please can you give us enough information to reproduce the problem you are seeing? If that's impossible (if you have an enormous system that isn't possible to share at the moment, for example), could you attach a commit log from the core plus the binary that it was running?

Without more information, I suspect we'll just have to close this issue because I can't see how we can make forward progress.

GregAC, 2020-04-06:
If it's not jumping back where you think it should one possibility is the `mepc` CSR has been written by software altering the point at which it should jump back.

If you are running this in simulation run it with tracing turned out and identify the point at which the jump to the exception handler happens, check the value of `mepc` in the CSR module and see if it's getting set as you expect. From this you can determine if there's some bug in the RTL causing an incorrect `mepc` to be set or something during the course of the exception handler than corrupts `mepc`.

tomroberts-lowrisc, 2020-04-06:
Can you share the ISR code? It looks like you might be returning from the interrupt handler with a regular function return, rather than using the ``mret`` instruction. If you are compiling with gcc, adding the interrupt attribute (``__attribute__((interrupt))``) to the ISR function will take care of this for you.

There is a simple example of this here:
https://github.com/lowRISC/ibex/blob/4f39160843ba1a6ad8186502af281e4aeba45e10/examples/sw/simple_system/common/simple_system_common.c#L185

zzyhotmail, 2020-04-08:
> Hi there! It's going to be quite hard for us to help you figure out what's going on from just this information. We think that our interrupt handling works correctly (and even have some tests for [...]

I'm tring to make a simple example to reproduce the issue.
Looks like it's going to take more time.

zzyhotmail, 2020-04-08:
> Can you share the ISR code? It looks like you might be returning from the interrupt handler with a regular function return, rather than using the `mret` instruction. If you are compiling with gcc, [...]

zzyhotmail, 2020-04-08:
> Can you share the ISR code? It looks like you might be returning from the interrupt handler with a regular function return, rather than using the `mret` instruction. If you are compiling with gcc, [...]

Yes, it is a ISR return near a regular function return.
I tried the __attribute__((interrupt)). It's seems that this method  solved the problem.
Thanks a lot.

closed 2020-04-08 by zzyhotmail