lowRISC/ibex #764: Introduce PMA handling to Ibex

GregAC, 2020-04-06:
In the OT repo we have been discussing whether we should trigger an alignment fault on any non-aligned access to device memory: https://github.com/lowRISC/opentitan/issues/1868

The best way to accomplish this would be a check in Ibex to see if something is attempting to perform an unaligned access to a device region and trigger an exception if so.

The RISC-V privileged spec allows use of PMA (physical memory attributes) where the properties of memory and what operations are allowable varies by memory address. Unlike PMP it doesn't specify how this should work leaving it up to implementors to decide.

I propose we add an PMA module to Ibex that would be given details of every memory access and be able to trigger faults. For the OT use case we may want something as simple as triggering an alignment fault on any misaligned access with the upper address bit set (or some other similar simple address check) so it needs to be able to implement that without adding extra overhead.

On the other hand we need to ensure other users of Ibex who may have more complex PMA requirements are able to implement them without major modifications.

The PMA module actually supplied in Ibex would be a barebones/no-op module which implementers would replace.

To perform DV we could provide a basic implementation so the OT use case gets tested or have the module implemented as a UVM agent and instead have it act 'randomly' to better test the ability of the PMA to implement arbitary policies.

I can sketch out more details but wanted some feedback on the general concept first.

rswarbrick, 2020-04-06:
How would this work from a timing point of view? Would using a complicated PMA imply another cycle of delay on memory accesses? Presumably we wouldn't want to add extra delay for the simple Ibex configuration. Would allowing both timing behaviours behind a parameter be difficult? (It sounds like it to me, but maybe I'm missing something!)

GregAC, 2020-04-06:
For the OT case I don't anticipate timing issues.

Certainly arbitrarily complex PMA could cause timing issues but this would be the implementers problem. For example they could choose to add a set of flops to the memory interface, though this has a bad impact on memory access latency. Or they could do something more complex, e.g. allow all transactions out onto the bus but flop anything headed for a device or any write, the PMA module could then kill the device/write transactions late after rule checking and for reads to normal memory (where you don't care about stray reads) just ignore the return data.

Whatever we come up with doesn't need to explicitly support such complex setups but be built with an eye towards making it easy for someone to implement such a thing.

GregAC, 2020-04-06:
There's also the question of whether this is implemented only the data interface or both the instruction and data interface. There's certainly a case for data side only as the instruction side already has its specific limitations (just reads, 32-bit always, may be 32-bit misaligned but always 16-bit aligned) and PMP can be used to prevent execution out of particular memory regions already.

tomroberts-lowrisc, 2020-04-06:
I think this would be a good idea. Should be much less pressure on timing than PMPs since everything will be pretty coarse-grained, and I would anticipate the configuration being synthesis-time rather than run-time.

asb, 2020-04-06:
The run-time vs synthesis time point is definitely worth clarifying. I agree that there seems to be no need for run-time configuration of PMAs - I can't imagine a case where that would be useful on an Ibex-class core outside of maybe enabling/disabling exceptions on misaligned access (which I belive the M3/M4 allow via a config register).

GregAC, 2020-04-06:
One way to implement this would be a function based rather than a module based approach. The function based approach would underscore the intention this is meant to be a synthesis time setup to hard-bake a memory map into the core rather than something that should be used for run-time configurable things.

rswarbrick, 2020-04-06:
That's a good point, but it might be a bit unpleasant from the repository organization point of view. Plugging in different modules into a Verilog design is reasonably standard; I'm not so sure about different functions... ?

In particular, you'd presumably want the function in a different file for each SoC layout. And you probably don't want the function in the global namespace. So I guess you put it in some named package which has several different definitions, one in each file? But then it looks awfully like a (combinatorial) module...

GregAC, 2020-04-06:
I think I'd favour the module approach, we can just have appropriate configuration/documentation that states we intended for this to implement a static layout rather than something dynamic and that a dynamic setup could lead to timing issues (i.e. Say to users: if you really want to, go for it, but on your head be it!).

tjaychen, 2020-06-22:
@GregAC 
@cdgori 
@sjgitty 
@moidx 

sorry this is a bit late, but if the PMA does go forward, could we consider additional attributes? For example if it's memory, scrambled or no scramble? This may ease downstream logic from building similar filters that check for the same thing.

In the case I'm thinking of, I wouldn't actually need the PMA to do anything internally, but just output an indication on the bus lines (if feasible) that the memory access i scrambled.

Specifically where this might be useful..the flash will likely have both scrambled and unscrambled regions, so the controller needs to know whether to apply de-scrambling upon reads. Instead of having another look-up downstream, it might be kind of nice to re-use this logic.  

Just a thought.

GregAC, 2020-06-23:
We could add some attribute output from the PMA (which could carry information such as whether the memory should be scrambled) though it will require modifications to the top-level IO to carry these bits as part of the memory requests. If it's something that's only useful for flash it may be better for the check to sit there rather than in Ibex. However if there's use for attributes being exported to the memory system for other reasons it could be a good feature to add.

tjaychen, 2020-06-23:
yeah we can definitely put it in the flash.  It's just that the check will be yet another "check address against range" to assign an attribute, so I was wondering if it made sense to put that functionality elsewhere if it already exists.

What do you think?

imphil, 2020-06-23:
We do have `user` bits in the OT interconnect to carry additional information, similar to the `user` bits in AXI. (We don't use them yet, however.)

tjaychen, 2020-06-23:
yeah that's kind of where i was going with this.  If we have PMA to do some kind of attribute assignment already, it makes sense to piggyback on that and use it for the rest of the system.  I am not super convinced myself yet, but just a thought.  I wanted to see what you guys and others think.

sjgitty, 2020-06-24:
ftr that's the type of functionality we had in mind by adding those `user`
bits to TLUL

cdgori, 2020-06-24:
@tjaychen seems reasonable to me to do this detection here from a location/reuse-of-logic standpoint.

But, basic question: What happens if this PMA-sourced attribute is glitched or miscalculated?  I can't picture all of the downstream consuming logic at the moment, but I assume that the result is that read data would not be unscrambled and therefore the contents would be garbage for the consumer.  A bit worse would be if supposedly-transparently-scrambled data makes its way to software (since it potentially facilitates cryptanalysis of the scrambling in a more direct way if somehow multiple consecutive reads can be seen like this, or a read and its scrambled form).  I honestly don't know if there is a worse-than-this scenario (but there could be) - could use your help to think about that.  If there is a big problem we might have to choose some encoding for the scramble/unscramble attribute to protect against this and consume >1 attribute bit to do so.

(Note: We still have this problem even if we compute the scramble/unscramble attribute at the target, but we would be able to have more redundancy in the structure/calculation there if merited based on the local usage of the attribute.  I think this just shows the need to evaluate the risk/threats across module/subsystem boundaries in cases like this.)

tjaychen, 2020-06-24:
@cdgori 
yes the worst case I had in mind is exactly what you said.  It would potentially enable the direct analysis of the scrambled contents by an attacker. I also agree that a multi-bit attribute would be best.

Some of the other complications that I hadn't mentioned is that the scrambling regions MAY need to be set by OTP; at a minimum they will need to be software lockable.  If we do need OTP entanglement, that may not work well with ibex, unless ibex wants to expose some ports to allow that.  It just does not feel very clean.