lowRISC/ibex #769: [DV] Run a full regression with failed cases returned

Fzl1029, 2020-04-08:
Hi

When I ran a full regression simulation with **ISS: spike** and **SIMULATOR: ius**. There are some failed cases as the following log. We can see that after full simulation, there are 295 Passed and **76 FAILED**.

[image]
From the first snippet of log, there exist some timeout errors as following:
[image]

First I suspected the failed case was caused by this kind of timeout error. But later I checked with regr.log and found that even the cases which have not problems with timeout error still have chance to fail. Thus I cannot figure out what causes those failed cases so far.

[image]
[image]

Besides, in the regr log, there also exist some errors like the following:
[image]

So my issue is: What causes these failed cases? Would you please help me figure out or give some clues for these failed cases? Is it acceptable or normal to have such number of errors? I have also attached with the two full log file I mentioned. Thank you for your help!

[build_ius_spike_full.log](https://github.com/lowRISC/ibex/files/4451758/build_ius_spike_full.log)
[regr.log](https://github.com/lowRISC/ibex/files/4451759/regr.log)

imphil, 2020-04-08:
IUS isn't a supported simulator (any more), as it doesn't support all necessary SystemVerilog language features. Can you retry with Xcelium or VCS?

GregAC, 2020-04-08:
I'd note even with a supported simulator the current regression isn't entirely clean @udinator I think you have recently run a full regression, any comments?

udinator, 2020-04-08:
Hi @Fzl1029 - Like @imphil says, we do not support IUS anymore for running simulations, and several of the tests do not pass with IUS.
Currently there are only a few tests that are failing, I'm looking into what is going on.

Fzl1029, 2020-04-08:
Thank you for all your help. I have noticed that you had removed all the ius option of your repository. So now I want to try with Xcelium. Which simulator option should I use?

Fzl1029, 2020-04-09:
Would you please also tell me you any specific information/details as to what feature should work with Xcelium that doesn't work with Incisive. Thank you.

udinator, 2020-04-09:
Hi, we do not currently support Xcelium for running simulations with ibex or running the random instruction generator (which is the first step of running a DV sim).
To add support for Xcelium, we would need to add another simulator target inside dv/uvm/core_ibex/yaml/rtl_simulation.yaml for Xcelium with any appropriate compile and runtime options, as well as adding a similar target in the google/riscv-dv repository, in the yaml/simulator.yaml file.
If you want, you can open up a PR to add support in this way, or I can look at adding support from my end, whichever works for you.

Re language features: the first error that caused compilation to fail deals with defining localparam variables upon declaration using the $clog2 function, which is unsupported in IUS.

imphil, 2020-11-02:
Xcelium support should be there now, I'm not sure how well it works, though. @Fzl1029 did you make any progress on that, or can you try again?

Fzl1029, 2020-11-02:
Hi, we have changed to use vcs after. If now ibex verification supports Xcelium again, I think we can give a try and let you know how well it works, thanks.

imphil, 2020-11-02:
Great! I've seen some recent bug reports about running the verification suite with XCelium, it would be especially interesting to see where VCS and Xcelium differ. If you find anything, could you please open a new issue to track it?

Let's close this one for now and look at new issues if they come up :-)

closed 2020-11-02 by imphil