lowRISC/ibex #776: RTL Simulation timeout in full regression

saurabhsingh-pvips, 2020-04-14:
Hi,

While running full regression using command `make WAVES=1 VERBOSE=1 ISS=spike COV=1` , RTL simulation gets multiple time outs of the following nature:

```
INFO     Running RTL simulation...
Tue, 14 Apr 2020 18:58:31 INFO     Command progress: 0/371
Tue, 14 Apr 2020 19:02:48 INFO     Command progress: 1/371
Tue, 14 Apr 2020 19:02:49 INFO     Command progress: 2/371
Tue, 14 Apr 2020 19:02:49 INFO     Command progress: 3/371
Tue, 14 Apr 2020 19:02:49 INFO     Command progress: 4/371
Tue, 14 Apr 2020 19:12:49 ERROR    Timeout[600s]:  env SIM_DIR=out/seed-26187/rtl_sim/riscv_pmp_test.0   out/seed-26187/rtl_sim/vcs_simv +vcs+lic+wait +signature_addr=8ffffffc  -ucli -do ./vcs.tcl  -cm line+tgl+assert+fsm+branch -cm_dir out/seed-26187/rtl_sim/test.vdb -cm_log /dev/null -assert nopostproc -cm_name test_riscv_pmp_test_0      +ntb_random_seed=26187 +UVM_TESTNAME=core_ibex_base_test +bin=out/seed-26187/instr_gen/asm_tests/riscv_pmp_test_0.bin     +ibex_tracer_file_base=out/seed-26187/rtl_sim/riscv_pmp_test.0/trace_core     -l out/seed-26187/rtl_sim/riscv_pmp_test.0/sim.log
```

One thing I figured out is that there is no file present at path specified here: `+ibex_tracer_file_base=out/seed-26187/rtl_sim/riscv_pmp_test.0/trace_core` .

GregAC, 2020-04-15:
The log extract you give indicates a timeout in the PMP test. Are all of the timeouts in a PMP test? 

What simulator logs are present? `trace_core_00000000.log` will only be generated if the Ibex core successfully starts running. The simulator log gives simulator output that may give clues as to why the test is timing out. They are named `sim.log` an example path from your log output is `out/seed-26187/rtl_sim/riscv_pmp_test.0/sim.log`

I believe the PMP tests are still under active development so may not be expected to work perfectly yet @udinator will know more.

You could also try with a longer timeout. This is hard-coded into `sim.py` you can see the lines that run the simulator here:

https://github.com/lowRISC/ibex/blob/5c51bc7e4d2db8d89f83ad57107961622b308e58/dv/uvm/core_ibex/sim.py#L267

and here:

https://github.com/lowRISC/ibex/blob/5c51bc7e4d2db8d89f83ad57107961622b308e58/dv/uvm/core_ibex/sim.py#L276

The first line is running via LSF (hence run_parallel_cmd) and the other without LSF. The 600/300 are the timeouts in seconds respectively. Your timeout of 600s from your log indicates you're running via LSF, is that expected? That could also be the source of the problem if the job isn't running correctly on LSF or times out awaiting a slot.

saurabhsingh-pvips, 2020-04-15:
The simulator log for `riscv_pmp_test` looks fine to me
```
Command: /home/saurabh/projects/rig/work/ibex/dv/uvm/core_ibex/out/seed-26187/rtl_sim/vcs_simv +vcs+lic+wait +signature_addr=8ffffffc -ucli -do ./vcs.tcl -cm line+tgl+assert+fsm+branch -cm_dir out/seed-26187/rtl_sim/test.vdb -cm_log /dev/null -assert nopostproc -cm_name test_riscv_pmp_test_0 +ntb_random_seed=26187 +UVM_TESTNAME=core_ibex_base_test +bin=out/seed-26187/instr_gen/asm_tests/riscv_pmp_test_0.bin +ibex_tracer_file_base=out/seed-26187/rtl_sim/riscv_pmp_test.0/trace_core -l out/seed-26187/rtl_sim/riscv_pmp_test.0/sim.log
Chronologic VCS simulator copyright 1991-2019
Contains Synopsys proprietary information.
Compiler version P-2019.06-SP2-3_Full64; Runtime version P-2019.06-SP2-3_Full64;  Apr 15 17:31 2020
```

Strange observation I made is that `riscv_pmp_test` doesn't cause issue when run alone. The timeout is only observed when regression is run. Does it still make sense to increase the timeout and try again?

GregAC, 2020-04-15:
Trying again with an increased timeout would be the next logical step, though if it's not producing a trace file that does indicate some issue that's stopping the test from running (rather than a test that's just taking longer than normal to run).

saurabhsingh-pvips, 2020-04-16:
Tried with 1200s timeout. Same error.

udinator, 2020-04-16:
Hi Saurabh, thanks for bringing this up.
I suspect that this timeout has something to do with the active-development status of PMP verification work for the time being; I have a rather large PR in the works on the riscv-dv side that when ready should hopefully fix these timeout errors.
I'll look more into what is going on tomorrow.
Are you running these simulations on LSF (or some equivalent)?

saurabhsingh-pvips, 2020-04-16:
Alright. Thanks.

No, I am not running on LSF.

udinator, 2020-04-18:
Hi @saurabhsingh-pvips, I have opened #797 to hopefully solve this issue of hanging PMP tests, I'll update this thread again to let you know once it gets merged.

udinator, 2020-04-20:
@saurabhsingh-pvips - could you try running the pmp test again? Several changes have been merged that should hopefully allow this test to run to completion.

saurabhsingh-pvips, 2020-04-21:
This still fails in regression but passes as individual test. Nothing fishy appears in sim.log

udinator, 2020-04-21:
I've identified the cause of this issue, and have opened #804 to fix it, will update once it gets merged.

udinator, 2020-05-01:
@saurabhsingh-pvips could you try this again? Some fixes have been submitted relating to how LSF commands are passed in that might fix this hanging error.

saurabhsingh-pvips, 2020-05-04:
Running on SPIKE gets the following error:
`error: bad --isa option rv32imcb`

Tried with OVPsim, test runs but time out before completion:
`ERROR    Timeout[10s] `
How do I increase the timeout for OVPsim? Added `--finishtime 1200` to `iss.yaml` but that doesn't work.

rswarbrick, 2020-05-04:
This is because we've recently added support for the bitmanip extension in Ibex. The master branch of Spike doesn't support that at the moment (probably because the extension hasn't yet been ratified). See the [Verification page](https://ibex-core.readthedocs.io/en/latest/verification.html) of the docs for instructions of how to install the version of Spike that you need.

saurabhsingh-pvips, 2020-05-04:
Tried with this version of spike: https://github.com/riscv/riscv-isa-sim/tree/riscv-bitmanip
Doesn't help. 
`error: bad --isa option 'rv32imcb'. unsupported extension 'b'`

vogelpi, 2020-05-04:
Hi @saurabhsingh-pvips ,
did you compile the `riscv-bitmanip`branch of Spilke with a compiler that actually has support for the bitmanip extension?

vogelpi, 2020-05-04:
+ when you call Spike, does it actually try to use the compiler with bitmanip support? I suspect that the error you see is because Spike tries to call the compiler that does not support bitmanip and thus does not recognize the flag.

saurabhsingh-pvips, 2020-05-04:
I'm using the lowrisc-toolchain via [here](https://github.com/lowRISC/lowrisc-toolchains/releases/tag/20200323-1).

I guess I am missing on applying bitmanip patch ([here](https://github.com/lowRISC/lowrisc-toolchains#how-to-generate-the-bitmanip-patches)). Can you confirm this?

imphil, 2020-05-04:
No, the toolchain you're referring to already contains all necessary patches to GCC and binutils to work with bitmanip. Spike is not part of the toolchain, you need to/did build it separately.

vogelpi, 2020-05-04:
I had a similar problem and it turned out that even though I had the new compiler installed, Spike was still using the old one because I did not change the `RISCV_TOOLCHAIN` environment variable.

saurabhsingh-pvips, 2020-05-05:
@imphil Yes, I did that. This response was to @vogelpi's query

> Also, when you call Spike, does it actually try to use the compiler with bitmanip support? I suspect that the error you see is because Spike tries to call the compiler that does not support bitmanip and thus does not recognize the flag.

saurabhsingh-pvips, 2020-05-05:
@vogelpi I don't have new installation of RISCV_TOOLCHAIN. I simply removed previous SPIKE with the one supporting bitmanip which is at the same previous location. This doesn't work.

vogelpi, 2020-05-05:
Hm, this seems to be a tough one. I am looking into this. It would be helpful if you could again:
- Post the exact command you are running from your shell (e.g. `make TEST=riscv_bitmanip_test ISS=spike ITERATIONS=1 VERBOSE=1`)
- Post the output you get.
You already did that when opening the issue, but I think the problem we now see is slightly different.
Thanks.

saurabhsingh-pvips, 2020-05-05:
Command: `make WAVES=1 VERBOSE=1 ISS=spike COV=1 SEED=26187` . Stuck to this seed since beginning of the issue I filed.

Output log: [full_regression.log](https://github.com/lowRISC/ibex/files/4579399/full_regression.log)
[riscv_arithmetic_basic_test.0.log](https://github.com/lowRISC/ibex/files/4579404/riscv_arithmetic_basic_test.0.log)

udinator, 2020-05-05:
Ah, it looks like the B extension is still being enabled, even though your version of Spike does not support it.
Inside the Makefile there is a variable named ISA which is set to 'rv32imcb', can you modify this to have value 'rv32imc'?

vogelpi, 2020-05-05:
Thanks @saurabhsingh-pvips for posting the command and logs. That is useful. Indeed it seems that your version of Spike does simply not recognize the ISA set to `rv32imcb`.
```
/home/saurabh/projects/rig/tools/lowrisc-toolchain-gcc-rv32imc-20200323-1/bin/spike --log-commits --isa=rv32imcb -l out/seed-26187/instr_gen/asm_tests/riscv_arithmetic_basic_test_0.o &> out/seed-26187/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.log
```
Could you please go to the directory (`riscv-isa-sim`) where you have downloaded the spike sources and do a `git show --summary` and then post the commit ID you get here?
Also, did you completely remove the `build` directory inside `riscv-isa-sim` and then rerun the configure step before running `make` and `make install`?

saurabhsingh-pvips, 2020-05-05:
@udinator
Did what you suggested. Error with riscv_bitmanip_test. 
``unrecognized opcode `sloi sp,zero,27'``

Find the log file here
[full_regression.log](https://github.com/lowRISC/ibex/files/4579744/full_regression.log)

vogelpi, 2020-05-05:
> @udinator
> Did what you suggested. Error with riscv_bitmanip_test.
> `` unrecognized opcode `sloi sp,zero,27' ``

This is supposed to fail as you now try to run a bitmanip test without specifying this to Spike. The specified architecture `rv32imc` does not have this instruction. To make it work, you have to exclude this `riscv_bitmanip_test` in the testlist: `ibex/dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml` (at the very bottom). This should do the trick for now.

But I think the cleaner solution would be to track down why your Spike version does not like `rv32imcb` ISA flag. I would suggest do checkout the `riscv-bitmanip` branch of `riscv-isa-sim`, remove the build directory, and rerun `make` + `make install`.

saurabhsingh-pvips, 2020-05-05:
@vogelpi My bad. I was working with incorrect commit. That's the issue. Checkout the right commit, performed the steps you suggested and it's working fine now. Made no changes to the `Makefile `or `ibex/dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml`

Will update on the issue originally filed.

vogelpi, 2020-05-05:
Excellent, I am glad it's working now. Thanks for your feedback!

saurabhsingh-pvips, 2020-05-05:
@udinator The timeout issue previously with PMP test is now coming for all the tests. Bumped timeout to 1200s but no success.
```
Tue, 05 May 2020 19:37:16 INFO     Running RTL simulation...
Tue, 05 May 2020 19:37:16 INFO     Running core_ibex_base_test with out/seed-26187/instr_gen/asm_tests/riscv_arithmetic_basic_test_0.bin
Tue, 05 May 2020 19:57:16 ERROR    Timeout[1200s]: env SIM_DIR=out/seed-26187/rtl_sim/riscv_arithmetic_basic_test.0   out/seed-26187/rtl_sim/vcs_simv +vcs+lic+wait +signature_addr=8ffffffc  -ucli -do ./vcs.tcl  -cm line+tgl+assert+fsm+branch -cm_dir out/seed-26187/rtl_sim/test.vdb -cm_log /dev/null -assert nopostproc -cm_name test_riscv_arithmetic_basic_test_0      +ntb_random_seed=26187 +UVM_TESTNAME=core_ibex_base_test +bin=out/seed-26187/instr_gen/asm_tests/riscv_arithmetic_basic_test_0.bin     +ibex_tracer_file_base=out/seed-26187/rtl_sim/riscv_arithmetic_basic_test.0/trace_core     -l out/seed-26187/rtl_sim/riscv_arithmetic_basic_test.0/sim.log
```
Timeout for all the first 5 `riscv_arithmetic_basic_test`.

> @saurabhsingh-pvips could you try this again? Some fixes have been submitted relating to how LSF commands are passed in that might fix this hanging error.

udinator, 2020-05-05:
@saurabhsingh-pvips - are you synced to the current top of tree? I'm unfortunately unable to reproduce this hanging issue from my end.

saurabhsingh-pvips, 2020-05-06:
I ran this on commit `fd01562ff7d46ceb6d368dc472c9305825ebcef5`.
Bumped to latest` b72d263eac25589fa1688f98d45deb2e3b147645` but same issue.

rswarbrick, 2020-05-06:
@saurabhsingh-pvips: I think Udi is most likely to be able to help with this, but I don't believe the current messages have enough information for him to reproduce what you're seeing. Since it seems you and I are on European time and Udi is in California, let's get a bit more information to avoid another day's delay.

Could you send a message with something like this:

> Running with Ibex commit <SHA> and a clean tree (use `git clean -fxd`), I ran the following command(s). They failed in the following way <short description here> and I'm attaching a log of what happened.

If you're using bash, I've found a good way to do this is something like this:

```bash
(set -x; git rev-parse HEAD; git clean -fxd; cmd0; ... ; cmdn) >commands.log 2>&1
```

then attach `commands.log` and any other log file to the issue.

saurabhsingh-pvips, 2020-05-06:
@rswarbrick Thanks for inputs.

@udinator 
Working on the commit `b72d263eac25589fa1688f98d45deb2e3b147645` and a clean tree. 
Command: `make WAVES=1 VERBOSE=1 ISS=spike COV=1 SEED=26187 > & full_regression.log`
Error: `ERROR    Timeout[300s]`
Find the log here:  [full_regression.log](https://github.com/lowRISC/ibex/files/4586181/full_regression.log)

Tried with timeout `1200s`. Same issue.

rswarbrick, 2020-05-06:
Right, I've just tried and failed to reproduce what you're seeing on the head of master (`7dafdf64`). I ran the following command (like you, but restricting the number of tests to make it a bit quicker to see what's going on):

     make WAVES=1 VERBOSE=1 ISS=spike COV=1 SEED=26187 TEST=riscv_arithmetic_basic_test 2>&1 | tee full_regression.log

I've attached the log file, which shows the tests running to completion. If you look at your log file closely, you'll see that each simulation run has its own log. For example, the one for the first failing test in your run is at `out/seed-26187/rtl_sim/riscv_arithmetic_basic_test.0/sim.log`. I've attached my version of that too. Please could you have a look at one of the simulation log files. If it's not obvious what's going on, could you send us that too?

[full_regression.log](https://github.com/lowRISC/ibex/files/4587006/full_regression.log)
[sim.log](https://github.com/lowRISC/ibex/files/4587011/sim.log)

saurabhsingh-pvips, 2020-05-06:
There is no `sim.log` generated for the command I ran. 

I have run into some VCS related issue at my end. Will try out running individual test instead of regression once that's fixed. Will update accordingly. 

@rswarbrick Can you please run a full regression as well? This happened with pmp_tests, it finished when run individually but timed-out in regression.
See below:
> The simulator log for `riscv_pmp_test` looks fine to me
>
> ```
> Command: /home/saurabh/projects/rig/work/ibex/dv/uvm/core_ibex/out/seed-26187/rtl_sim/vcs_simv +vcs+lic+wait +signature_addr=8ffffffc [...]

rswarbrick, 2020-05-06:
@saurabhsingh-pvips As far as I understand the conversation above, Udi thinks everything is fixed. I assume that he came back to you to check that you don't have anything weird in your environment differing from his. There's not much point in me spending time trying to reproduce a bug that you may or may not have.

Please try to reproduce the problem you are seeing. If you can reproduce it, please send logs showing what's going on. If you can't, please close the issue.

vogelpi, 2020-07-03:
There has been no activity in this issue for almost two months now. I am thus closing it. Feel free to re-open it if the problem still persists.

closed 2020-07-03 by vogelpi