lowRISC/ibex #778: [dv] Add clocking blocks to icache UVM code

rswarbrick, 2020-04-15:
See discussion on PR #761, in particular [these comments](https://github.com/lowRISC/ibex/pull/761#discussion_r405777417).

Sri and Udi have suggested that the code would be better if the interface included clocking blocks. Suggested code:
```systemverilog
clocking driver_cb @(posedge clk_i);
  output req;
  output branch;
  output branch_addr;
  output ready;
  output enable;
  output invalidate;
endclocking

clocking monitor_cb @(posedge clk_i);
  input valid;
  input rdata;
  input addr;
  input err;
  input err_plus_2;
  input busy;
endclocking
```

rswarbrick, 2020-04-15:
For example, some code in the driver currently looks like this:
``` systemverilog
    // Assert ready and then wait until valid
    cfg.vif.ready <= 1'b1;
    while (1'b1) begin
      @(posedge cfg.vif.clk);
      if (cfg.vif.valid)
        break;
    end
```
This reads from `valid` (driven by the DUT, so in `monitor_cb` above) and writes to `ready`. They share a clock. Is two clocking blocks the right way to structure this? I'd appreciate some guidance!

rswarbrick, 2020-04-15:
@udinator, @sriyerg: Could you give me some advice about how this should look?

rswarbrick, 2020-04-15:
Reading [Cummings's paper](http://www.sunburst-design.com/papers/CummingsSNUG2006Boston_SystemVerilog_Events.pdf) more carefully, I think I have an idea of how this would work.

Use a single clocking block, specifying `default output negedge` to make it more obvious what's going on in dumped waves. The code above would be:

``` systemverilog
cfg.vif.cb.ready = 1'b1;
while (1'b1) begin
  @(cfg.vif.cb);
  if (cfg.vif.cb.valid)
    break;
end
```

Suppose that there's a clock with a posedge every 100ns and let's pretend that `#1step` is 1ns. Then I think a sample execution would look like

Time (ns) | Event
---|--
0 | `cfg.vif.cb.ready = 1'b1` executed
51 | ready = 1 actually happens
... | ...
700 | valid = 1 (set by an `@(posedge clk)` block in DUT)
799 | valid sampled by cb (`#1step` before time step at 800ns)
800 | break

Does that make sense? If so, I propose to use a single clocking block and I'll push up a PR tomorrow. If not, can you let me know what would be better?

rswarbrick, 2020-04-15:
Also, should this be one clocking block with `inout` items, or two clocking blocks, one for the driver and one for the monitor? The latter means you need fewer `inout`s, but it feels a bit icky because the interface class then gets some monitor-specific and driver-specific code. Or is there another trick possible?

udinator, 2020-04-16:
I'll defer to Sri on the usage of `default output negedge`, I'm not quite sure how necessary this would be here.
After looking at this a bit more, I think that using a single clocking block would be fine as well, I don't think we would need any `inout` signals here. If you want to enforce different timing requirements on signals in the "monitor" and "driver" set, then it's probably better to use two clocking blocks (moving `valid` into the driver block since it's part of the driver protocol) to have some separation between the two sets of signals, otherwise I'd be fine with just a single block.
@sriyerg what do you think?

sriyerg, 2020-04-16:
- There is any need to add extra skews to the clocking block, unless the DUT is timing sensitive.
- Prefer N+1 clocking blocks (at least), where N is the number of flavors of drivers the agent has, and 1 for the monitor. So just 2 in your case.
- Refrain from `inouts` - the purpose is to also make the port list view clear (see TileLink interface example below with 2 flavors of drivers, to mimic host or a device, in addition to the monitor's view, which is input-only). 
- I wouldn't consider clocking block as being driver code or monitor code specific, since they are just providing a view of the port list.
- This is a very common way of writing SV interfaces.  
- Here's the TileLink interface as an example - 
   https://github.com/lowRISC/opentitan/blob/master/hw/dv/sv/tl_agent/tl_if.sv
- If you go one level up, there are some other agents as well. 

The two clocking blocks would look something like this:
 ```systemverilog
  // Used exclusively by the driver to access the port list it is driving and sampling. 
  clocking driver_cb @(posedge clk);
    output req;
    output branch;
    output branch_addr;
    output ready;
    input  valid;
    input  rdata;
    input  addr;
    input  err;
    input  err_plus2;
    output enable;
    output invalidate;
    input  busy;
  endclocking

  // Used exclusively by the monitor to access the port list it is sampling.
  clocking monitor_cb @(posedge clk);
    input  req;
    input  branch;
    input  branch_addr;
    input  ready;
    input  valid;
    input  rdata;
    input  addr;
    input  err;
    input  err_plus2;
    input  enable;
    input  invalidate;
    input  busy;
  endclocking
```

Yes, the full set of signals is replicated in both clocking blocks, because again, its just a view of the port list (what ports are allowed to be driven and what ports are allowed to be sampled from that clocking block). By limiting the driver and monitor to only refer to signals from clocking blocks, you also get that implicit check (for example, if driver drives an input signal accidentally, it will throw an error at compile-time).

rswarbrick, 2020-04-16:
Right, understood. I'll write something for the core agent today and push it up.

rswarbrick, 2020-05-12:
Oops, this got done with #786.

closed 2020-05-12 by rswarbrick