lowRISC/ibex #780: Long path through multdiv into IF stage

vogelpi, 2020-04-15:
During his Zbt work #746 @ganoam noted that there is a long path from the register_file -> *multdiv* -> alu -> ex_block/branch_decision_o -> controller -> IF stage.

The strange thing is that while the path through multdiv and alu is expected (we use the adder for divisions), the combined path should never end up in the controller, branch decision, IF (these ops only depends on the alu). So my suggestion was to qualify the branch decision in the controller with an early signal (the branch decision should never matter if we are doing a multdiv). But this did not really help.

Maybe @tomroberts-lowrisc or @GregAC have some ideas?

The timing report can be found [here](https://github.com/ganoam/ibex/blob/bitmanip-devel-zbt/reports/bitmanip-zbt-pr/33bitshifter_adderbinmux_optimized/RV32B0_3stage_btalu_2_ns/post_synth_timing.rpt).

tomroberts-lowrisc, 2020-04-15:
I guess the branch decision always depends on the output of the adder. The inputs to the adder are always muxed between various sources including the multdiv operands. The multdiv operand b is always muxed between the raw input sources (including the register file).

I think there's not much that can be done about that other than re-arranging and combining the operand muxes in multdiv and the id stage.

GregAC, 2020-04-15:
(Tom posted this just before I hit the comment button so more or less repeating what he said)

The issue here is the worst path going into the ALU comes via the multdiv. Qualifying the signal coming out of the ALU that it doesn't matter if the operands have come from the multidiv doesn't effect the timing path (in terms of worst case propagation time through the physical gates, your qualification adds to the circuit, it hasn't broken a path that did exist).

You can tell your timing analysis that this is a false path (i.e. you don't care that fails timing as it'll never actually cause a problem) but generally such things are to be avoided as they can cause problems (e.g. you mess up the RTL changes that ensure the false path doesn't matter, such as the qualification you suggested, as they don't really get tested unless you do a full gate-level simulation with back annotated timing info).

One fix would be an entirely separate branch condition unit (or a dedicated ALU for the multdiv), with the obvious area cost. Otherwise we need to look at ways to make those multdiv operands earlier or improve the path on the output side of the ALU.

vogelpi, 2020-04-16:
Thanks for outlining the options guys. I think re-arranging and combining the operand muxes to make the operands earlier and improve the paths outside of the ALU is the most reasonable, but it will require a lot of work. I think we should leave it for the moment. Maybe that's something to do once the design is super stable.