lowRISC/ibex #784: Add AscentLint lint target to Ibex

imphil, 2020-04-16:
Currently we have a verible and a Verilator lint target in Ibex, and an AscentLint lint target in OpenTitan. Each target has an associated waiver file. For consistency, we should think about adding the AscentLint target to Ibex as well, together with the waiver file that currently lives at https://github.com/lowRISC/opentitan/blob/master/hw/ip/rv_core_ibex/lint/rv_core_ibex.waiver

We cannot run AscentLint in CI on Ibex, but we'll get the lint results by either manually running the tool (if someone has access to it), or as soon as Ibex is vendored into OT.

imphil, 2020-04-16:
@msfschaffner sounds like a plan?

msfschaffner, 2020-04-17:
Sounds good! Are you going to move that waiver as part of #785? If so, just send me a draft PR on either OT or the Ibex repo and I'll double check whether it works correctly.