lowRISC/ibex #785: Get Ibex AscentLint-clean

imphil, 2020-04-16:
Go through https://reports.opentitan.org/hw/ip/rv_core_ibex/lint/ascentlint/latest/results.html and try to understand/fix all lint errors. Talk to @msfschaffner to waive the ones we don't care about (the waiver lives in OpenTitan at https://github.com/lowRISC/opentitan/blob/master/hw/ip/rv_core_ibex/lint/rv_core_ibex.waiver).

(We could think about moving the AscentLint waiver into the Ibex repo as well, I filed #784 for that.)

rswarbrick, 2020-04-16:
Looking at these warnings, lots of them are also reported by our existing Verible lint flow already. Maybe we should get that clean first, since that will probably fix a good proportion (without needing to go via someone with an AscentLint licence).

msfschaffner, 2020-04-17:
Sounds good. If you need more timely feedback than what the dashboard provides, just make a draft PR and PM me, and I'll run it for you.

vogelpi, 2021-04-12:
I think we can close this issue now as Ibex is indeed AscentLint-clean (at the moment). WDYT @imphil ?

rswarbrick, 2021-05-21:
Looks good now: https://reports.opentitan.org/hw/ip/rv_core_ibex/lint/ascentlint/2021.05.20_23.08.13/results.html :-)

closed 2021-05-21 by rswarbrick

msfschaffner, 2021-05-21:
Awesome!