lowRISC/ibex #790: Verilator memutil: Support 64 bit wide memories

imphil, 2020-04-17:
Currently memutil expects the input file (ELF/BIN) to be 32 bit words, and the memory also to be 32 bit words. We need to support the case where the input file stays 32 bit wide, but the memory is a multiple of that, e.g. 64 bit.

https://github.com/lowRISC/ibex/blob/master/dv/verilator/memutil/cpp/verilator_memutil.cc#L426

https://github.com/lowRISC/ibex/blob/master/shared/rtl/prim_generic_ram_1p.sv#L93

This change is also needed in OT. First, do a PR in Ibex. Then copy do a PR in OT, updating Ibex through the vendor tool, and the primitive separately.

tomroberts-lowrisc, 2020-04-17:
I was initially going to error out if Width wasn't a multiple of 32, but actually we probably want to support that case for ECC/parity. Would need to load the data in 32bit chunks but leave any upper bits untouched (assuming zeros are ok for whatever scheme is being used). Any thoughts on that @imphil ?

imphil, 2020-04-17:
I'd go for error out at the moment (on the memutil side) until we have a clear picture of how we want to pre-process or pad data which doesn't fit in nicely. I think Tim has some ideas/requirements for a future iteration, but I don't know them yet.

PR #791 (2020-04-20): [memutils] Add support for > 32b memories
Pack > 32bit memory loads into wider RAM instances. Fixes #790
commit dc8ba83d (2020-04-20): [memutils] Add support for > 32b memories

Pack > 32bit memory loads into wider RAM instances. Fixes #790
closed 2020-04-20 by tomroberts-lowrisc