lowRISC/ibex #802: How to run risc_instr_cov_test for coverage generation during rtl_simulation?

saurabhsingh-pvips, 2020-04-21:
As per my understanding, the coverage generation is done by script `cov.py` the following way:

[image]

This is zero time coverage simulation generated from trace.

I wish to have coverage generated during rtl simulation. But I am unable to run the `riscv_instr_cov_test` in `riscv_dv_extension/cov_testlist.yaml`.

I think there should be an opt for cov_test in `riscv_dv_extension/testlist.yaml` similar to `gen_test` & `rtl_test`. But I am not able to implement this.

Is my approach correct or am I missing out on something ?

udinator, 2020-04-21:
Hi @saurabhsingh-pvips - good question, I think there are actually two things being mixed here:

1) To generate code coverage during the RTL simulation itself, add the `COV=1` argument to your Makefile command, and this will generate a `test.vdb` coverage database into `<output_dir>/rtl_sim/` that you can then access. The `cov` target in the Makefile is in charge of simulation-time code coverage generation, and will automatically be invoked by the flow, as said before all that you need to do from your side is add `COV=1` to the Makefile command you run.
2) The `fcov` target that you mention above is what any flow integrated with riscv-dv generation will refer to as the "functional coverage" generation step.  This target should be invoked manually from the command line *after* RTL simulation and post comparison is completed (e.g. `make SEED=123 TEST=riscv_rand_instr_test fcov`), making sure to explicitly specify the correct intended seed otherwise this will produce incorrect results.
At high level, what this step does is to parse all created Ibex tracer log files and analyze all the types of executed instructions and their operands to determine (based on coverpoints within riscv-dv itself) what has been covered from the perspective of the generated instructions and operands (e.g. if it is seen that not enough covergroups relating to load and store instructions and operands have been hit, add more iterations/types of memory access tests, etc...).

Does this clear up the situation?

saurabhsingh-pvips, 2020-04-22:
Yes. So this `fcov` is what I've been missing. I'm able to generate Functional Coverage.

But this `fcov` is still zero time simulation right? Things like transition bins are not going to be covered. Any inputs or hints on how to achieve transition coverage?

udinator, 2020-04-22:
I think if I am understanding your question about transition coverage correctly, you are referring to coverage of small sequences of instructions that share source/destination registers? This is something that is still a WIP on riscv-dv side, as being able to cover longer and longer sequences of 
instructions will create a huge number of coverage bins to create.

The directory structure is actually set up to output each regression/simulation in a separate directory corresponding to its global seed, to better separate different simulation runs. For example in the case of your simulation, all RTL simulation log files and related trace logs will be in `out/seed-26100/rtl_sim`, and the generated instructions and compiled binaries will be in `out/seed-261000/instr_gen`

Hmm, thanks for raising this interesting issue with the unwanted `uvm_fatal`, I will take a closer look at what's causing this tomorrow.

saurabhsingh-pvips, 2020-04-22:
Yes. That is exactly I wished to asked. Let's say I wish to cover a scenario in which register say x20 is accessed by an ALU instruction and immediately followed by Load-Store instruction or such. Is it possible to achieve this by analyzing the trace cores or there is need for some other approach? What way would you suggest if I wish to implement such coverage model on my own?

udinator, 2020-04-23:
Yep, we do actually already provide coverage for this sort of scenario: we support all sorts of data hazards that result from a sequence of two instructions, sequences of 3+ instructions is still an action item and has not been implemented yet.
For context, you can see here: https://github.com/google/riscv-dv/blob/master/src/riscv_instr_cov_item.sv#L231, which parses for data hazards.

udinator, 2020-05-05:
@saurabhsingh-pvips - the `uvm_fatal` issue you brought up has been fixed.

closed 2020-05-06 by udinator