lowRISC/ibex #808: SpecialReqAllGivesSpecialReqBranchIfBranchInst assertion failure

udinator, 2020-04-23:
Hi all,
When running the `riscv_mmu_stress_test` with PMP regions enabled, I noticed that the `SpecialReqAllGivesSpecialReqBranchIfBranchInst` assertion was failing, shown below:
[image]
A snippet of the OVPsim log of the relevant instruction execution is shown below:
[image]
A waveform snippet of when the assertion fails is below, the failure occurs one cycle after the yellow marker, where the small red arrow is located:
[image]
The situation is as follows: the `sb` instruction at `PC=0x80005496` (at the green `M1` marker accesses memory at `0xffa839bc` from User-mode and this access correctly causes a PMP store address exception (all 16 regions are enabled and set to `TOR` mode, the address being accessed here is not contained within any of the enabled regions).  The core then jumps to the PC stored in `mtvec`, which is a jump instruction at `PC=0x80001400`.
However, one cycle after starting to execute this jump, the previously mentioned assertion fails, as even though `special_req_all` and `jump_set_i` are both high due to the PMP store exception and the resultant branch, the `special_req_branch` signal does not go high at all.

tomroberts-lowrisc, 2020-04-23:
Looks like the pmp error isn't being cleared correctly in the lsu - I'll take a look

tomroberts-lowrisc, 2020-04-24:
@udinator - hopefully now fixed but leaving the issue open until you've had a chance to rerun the test.

udinator, 2020-04-24:
This seems to be fixed now with your change - thanks!
I'll go ahead and close this issue.

closed 2020-04-24 by udinator