lowRISC/ibex #826: [bitmanip] [zbe] bit deposit and bit extract (bdep/bext) instructions

ganoam, 2020-04-29:
The zbe instruction group implements the generic bit extract and bit deposit functions. `bext` collects LSB justified bits to rd from rs1 using extract mask in rs2. `bdep` writes LSB justifie dbits from rs1 to rd using deposit mask in rs2. 

The implementation is planned to base upon the butterfly network as described in [Hilewitz et al.](http://palms.ee.princeton.edu/system/files/Hilewitz_JSPS_08.pdf). 

Similar reference implemnetations by Claire Wolf indicate a rather large increase in area caused by this architecture. There are several possibilities to trade some latency for a smaller core, but the benefits are limited. However, in a complete RV32B implementation there are several functionalities, which could share the butterfly network, and therefore somewhat mitigate the area increase. For example it should be possible, to reuse zbe structures for bitcount instructions (ctz, clz, pcnt), single-bit instructions (zbs), bit ppermutation (zbp), pack, rotate and cmix instructions. 

The plan is to first implement the butterfly network just for bdep and bext. At that point we will see how bad the area increase is. I will then try to incorporate as many of the existing instructions as possible into the structure.

vogelpi, 2020-04-29:
This plan makes sense to me @ganoam . It will be very interesting to see how much can be shared with other instructions.

ganoam, 2020-05-12:
PR #863 is ready.

ganoam, 2020-05-22:
This has been merged.

closed 2020-05-22 by ganoam