lowRISC/ibex #832: PMP load error qualifies register writeback

udinator, 2020-04-30:
Hi,
I encountered this situation when running some pmp tests on Ibex:
[image]
The instruction at `PC=0x80002250` is an `lb` instruction that tries to access the address `0x800148e4`. One cycle after this instruction reaches the decode stage, it causes a PMP fault and `pmp_err_q` goes high for two cycles, but `lsu_rdata_valid_o` still goes high to `1'b1` even though the DMEM interface has not sent out a request at all, triggering an incorrect register writeback of of `XXXX_XXXX`.

tomroberts-lowrisc, 2020-05-01:
Hi Udi
I can see the issue here, but it looks like all memory errors are a bit broken. When I run an instance of riscv_mem_error_test, I see every load that gets an error (data error not PMP error) is writing back data to the register file, but the tests are passing? I guess it only fails here because the data is X rather than garbage.

I'll get a PR ready with the fix, but it would be useful to investigate what's going on with the memory error tests.
Cheers
Tom

udinator, 2020-05-01:
Yep, I just checked and you're right that register writes on load errors are still being performed as well -  looks like this slipped through because of how ISS log comparison is done for this test. On the mem_error_test , only the final set of register values are cross-checked against the ISS log for correctness as this test involves external stimulus (which it is currently impossible to get the ISS to concurrently simulate) so any intermediary register writes do not get checked with the ISS log, but on my PMP tests, every instruction is checked against the ISS log since they natively support PMP functionality, which I believe is why this case was caught.

PR #835 (2020-05-05): [rtl] Stop regfile writeback for load errors
- A data or PMP error should stop the register file from being updated
- Fixes #832
commit 15ab023e (2020-05-05): [rtl] Stop regfile writeback for load errors

- A data or PMP error should stop the register file from being updated
- Fixes #832
closed 2020-05-05 by tomroberts-lowrisc