lowRISC/ibex #840: The contents of dv/uvm/data are not vendored properly from OpenTitan

rswarbrick, 2020-05-04:
These were copied from OpenTitan as part of the mega-commit at PR #711. Since they are copied into the main Ibex repo, rather than vendored in, they don't stay in-sync with changes there.

Recently (30th March), some of the keys for coverage generation changed (see OpenTitan commit 39ffebd). These don't now match and you can't run `dvsim.py` to generate coverage without hacking `dv/uvm/data/vcs/vcs.hjson`, at least.

I think we should probably be vendoring this in properly. Is it part of dvsim? In which case, maybe it needs moving around in the OT repo and getting pulled in with the existing rules. If not, we need (yet) another vendor rule and either way this code needs to live under `/vendor`.

udinator, 2020-05-04:
The files in `dv/uvm/data` are needed for dvsim, to specify project-wide variables and default values; same with the contents of `dv/uvm/data/vcs` as well.
It is definitely possible to vendor these into the Ibex repository - this will result in 3 additional vendor rules:
1) The `dv/uvm/data/common_project_cfg.hjson` in the Ibex repo lives at `hw/data/common_project_cfg.hjson` in OpenTitan.
2) The 4 `.hjson` files, `sim.mk`, `vcs/vcs.hjson`, and `xcelium/xcelium.hjson` that are all currently under Ibex's `dv/uvm/data` directory all live at `hw/dv/data` in OpenTitan.
3) `cover.cfg`, `vcs.fsdb`, and `xprop.cfg` that are currently under Ibex's `dv/uvm/data/vcs` directory all live at `hw/dv/tools/vcs` in OpenTitan.

It is also likely that by vendoring in all of these files, we will need to patch them (and update the patches) somewhat regularly to fix directory paths and things like this that will keep being added onto these configuration files.
WDYT?

imphil, 2020-05-04:
Vendoring is probably only half of a solution here.

Ideally, the dvsim tool would work with less repeated config -- all default config should be part of the tool itself (which we vendor in), and only deviations from that should be in our config files. That's especially true for all files @udinator mentioned in (2) above (e.g. `sim.mk`).

I'd take that as a medium-term goal, and for now simply keep the two configurations manually in sync. A git diff between the two data directories in OT and Ibex shows some deviations which seem to be intentional, and others which aren't. Maybe @udinator you can align them again as much as possible, and open a tracking issue to get this sorted on the dvsim side?

udinator, 2020-05-05:
@imphil - I definitely agree, this warrants a look at the organization of dvsim default config files to try and consolidate them.
For the time being, I'll get a PR open to update the existing files in `dv/uvm/data`.

rswarbrick, 2020-05-06:
I've just pushed up https://github.com/lowRISC/opentitan/pull/2141, which tidies up the various filenames enough that we can vendor in the `hw/dv/data` and `hw/dv/tools` directories with minimal patching. (I tried it locally and it works!)

udinator, 2020-05-07:
Sounds good! Let's wait until that PR gets merged, and then we can vendor in those directories.

rswarbrick, 2021-01-25:
This was done by #1201.

closed 2021-01-25 by rswarbrick