lowRISC/ibex #843: it's confused about the user_extension.svh, there should be a better way

gxflying, 2020-05-06:
hi, when I try to use the extension facility of the riscv_dv_extension, I find there is a sv header file named "user_extension.svh" , which already exists under the origin ./google_riscv-dv/user_extension/.

actually it gets the right result when ”make gen” and from the compile log , we can see that the file "user_extension.svh " under ./riscv_dv_extension/ is the file really included and is used.  I think this is because of the parsing rule of the vcs.  the "+incdir+riscv_dv_extension" option in the command line is ahead of the option "+incdir+xxxxxxx/google_riscv-dv/user_extension". 
[image]

but I think we should NOT depend on the parsing rule to ensure the correctness of the user extension facility and it can be confused. 
actually the google dv system already supports the user-defined extension source files in user-defined path by the "--user_extension_dir" option of the run.py . 
other than including two extension path(google defined & ibex defined), this option will replace the path "./google_riscv-dv/user_extension/" with the user specified extension which is more clear and is a better way in my a pinion.
[image]

rswarbrick, 2021-01-27:
@udinator: I'm not completely sure I understand this report, but should it actually be filed against riscv-dv?

udinator, 2021-01-28:
@gxflying I'm not quite sure I understand your concern but let me try to answer.
We use the `user_extension.svh` to include an Ibex-specific class extension from riscv-dv, and then perform a type override in the `Makefile` in the `gen` target.
We also include the `google_riscv-dv/user_extension` directory to allow users to freely experiment with modifications to Ibex and the instruction generator.
If I've not answered your original question please feel free to correct me.

rswarbrick, 2021-04-05:
I'm not sure that there's anything left to do here. @gxflying: please re-open (and maybe explain a bit further) if you think there's something that should change in Ibex.

closed 2021-04-05 by rswarbrick