lowRISC/ibex #849: PMP store fault does not jump to exception handler

udinator, 2020-05-07:
Hi Tom,
I ran into this situation when debugging a PMP test (with icache disabled):
[image]
The instruction at `PC=0x8001fb0a` (under the cursor) is a store instruction to address `0x80031034`, which the PMP detection logic correctly flags as an error for not matching any pmp_cfg CSR in user-mode and prevents a request from being sent to the memory.
During this, the controller is in the `DECODE` state, but it does not actually detect `special_req_all` going high because `instr_valid_i` has already gone low by the time it reflects the rising `exc_req_lsu` wire - as a result the core pretends everything is okay and goes on to execute the instruction at the next PC.
I suspect this has to do with how `instr_valid_id_d/q` are set in the IF stage, as they appear to be dropping one cycle too early - additionally, when I enabled the icache and reran this test, this issue did not appear, e.g. `instr_valid_i` in the controller went high at the same time as `special_req_all` to qualify it due to `instr_valid_id_d/q` staying high for an additional cycle:
[image]
We can see that with the icache, `instr_valid_id_d/q` stay high for extra cycles, allowing the controller to correctly qualify the PMP exception and then start fetching the address of the exception handler (`0x80001400`).

tomroberts-lowrisc, 2020-05-11:
Thanks for the report Udi - I'll push a fix shortly.
The instr_valid_i signal drops because the store instruction has moved to the Writeback stage, and in this case there is no other instruction behind it (due to IF stage stalls). The icache probably manages to get another instruction in behind the faulting store here, but we should still be able to hit the same bug eventually.
I'm not sure this is unique to pmp errors though. We should be able to hit this same issue with memory errors?

udinator, 2020-05-11:
Hi Tom, I agree, I was also not completely sure this applied only to PMP errors, but (so far) I haven't been able to recreate this error with the memory error test.
I'll give your pull request a try and see if it fixes this issue, thanks!

PR #854 (2020-05-15): [rtl] Fix instr_valid_i exception issue
- The controller state machine could only progress to FLUSH to handle an
  exception if instr_valid_i was set
- When the exception comes from a load/store in the Writeback stage, and
  no new instruction has been driven into the ID stage, this could cause
  exception to be missed
- The instr_valid_i qualification is therefore removed from the state
  machine as all relevant signals inside that if block are already
  qualified by instr_valid_i anyway
- Fixes #849
commit 8934267c (2020-05-15): [rtl] Fix instr_valid_i exception issue

- The controller state machine could only progress to FLUSH to handle an
  exception if instr_valid_i was set
- When the exception comes from a load/store in the Writeback stage, and
  no new instruction has been driven into the ID stage, this could cause
  exception to be missed
- The instr_valid_i qualification is therefore removed from the state
  machine as all relevant signals inside that if block are already
  qualified by instr_valid_i anyway
- Fixes #849
closed 2020-05-15 by tomroberts-lowrisc