lowRISC/ibex #852: 'VERBOSE' mode doesn't give any extra information when running a single test

tudortimi, 2020-05-09:
I was having an issue running a simulation using OVPSim. It wasn't dumping anything in the log file and the comparison was failing. I tried running with `make ... VERBOSE=1`, hoping that it will give me some more log output to figure out what the issue was. The output was the same.

In the end, I solved my issue using good old fashioned print debugging (https://github.com/google/riscv-dv/issues/565).

rswarbrick, 2020-05-11:
Hi @tudortimi. Thanks for the message. The VERBOSE argument here (in `dv/uvm/core_ibex/Makefile`) turns into a `--verbose` argument to `dv/uvm/core_ibex/sim.py` or the vendored riscv-dv `run.py`.

Looking at the linked issue, I guess you were expecting more information when running OVPSim, which is done by `run.py`, which is vendored from riscv-dv. Please could you open an issue there, explaining what information should be printed? Then we can solve the problem here by fixing that issue, then vendoring in riscv-dv with the fix.

tudortimi, 2020-05-11:
You mean open a case on `riscv-dv` for what I would like their script to print when called with `--verbose`, right? I can do that.

tudortimi, 2020-05-11:
Additionally, I think it would be very helpful if VERBOSE mode (maybe with different selectable levels) would print more information about the steps it takes to execute, for example what external scripts are called and with what arguments, to assist in debugging.

In this concrete case, with the OVPSim issue, if `VERBOSE=1` would have echoed the commands being called in the Makefile, it would have only brought me so far as to know that the issue is in the script run here: https://github.com/lowRISC/ibex/blob/863fb56eb166dc8d6e25f4e98b7319f0c7088564/dv/uvm/core_ibex/Makefile#L275

It's much easier for someone new to follow what's being executed if the commands are shown in the console.

rswarbrick, 2020-05-11:
Yes, I agree. I think this was a design decision made by the original authors. For what it's worth, `make --trace` disables the '@' symbols, which can help with this sort of thing.

tudortimi, 2020-05-12:
It's slightly awkward that there are two ways to configure the verbosity. Would you be open to a PR with a Python script that wraps the Makefile? I find such scripts to be better also because it's possible to get a good overview using the `--help` switch.

rswarbrick, 2020-05-12:
I don't think that's a good idea, personally. I would, however, be keen on a PR that makes all the '@' signs in the Makefile conditional on the VERBOSE variable. This could work the same way as, say, Automake handles the V=0/V=1 parameter.

rswarbrick, 2020-05-12:
@tudortimi: Actually, I realised this was going to be really easy, and something I wanted too, so I've just done it. If the PR gets merged, do you think there's anything else that needs tracking on this issue?

tudortimi, 2020-05-12:
I think this is enough for now. If there is anything else, I can always open a new issue.

closed 2020-05-12 by rswarbrick