lowRISC/ibex #853: test "riscv_dret_test" format error in testlist.yaml terminats the command line

gxflying, 2020-05-11:
Hi, i find a typo in the testlist.yaml. details below:
there is a **unexpected space line** in the **sim_opts** node, which will terminate the command line.
and the cmd fails to generate a .S file
[image]

rswarbrick, 2020-05-11:
Hi @gxflying. Can you give the exact SHA of the tree where you see this? Looking at `863fb56e`, I see an empty line 225, which is what I would expect.

gxflying, 2020-05-11:
I use the lastest verson of the master branch.
it seems good that there is a space line between two test node.
but when the difference between the two syntax below, the space line may not behavior as you think. the
1.  sim_opts: >
2.  sim_opts:
in my situation, 1->error 2->ok
or you may have a try by executing "make TEST=riscv_dret_test gen" command line and see if the assembly file generated.

rswarbrick, 2020-05-11:
I'm sorry, I think you'll have to send more details. I just ran:
```
make -C dv/uvm/core_ibex TEST=riscv_dret_test gen
```
from the top of the Ibex tree and 5 assembly files are generated in `dv/uvm/core_ibex/out/seed-24833/instr_gen/asm_tests`.

As I asked above, please can you give the exact SHA of the tree where you see this? Please also check that you haven't got any uncommitted changes in your tree (run `git status`). If the test is still failing for you, please also attach the log.

gxflying, 2020-05-11:
ok, I'll check it and do some more test tomorrow,  any results I'll inform you. thanks

udinator, 2020-05-11:
Hi @gxflying - this is an interesting issue.
The `sim_opts` node you mentioned is actually unnecessary for instruction generation, as this only impacts the RTL simulation step of the Ibex flow, only the `gen_opts` node has any effect on instruction generation.
Something else I noticed was that in the snippet you posted containing the command, the `+require_signature_addr=1` parameter appears twice - it should in theory only show up a single time in the command. For example, when I generate a dret test, my command looks like the following:
[image]
You'll see that there are some extra spaces in this command, but this should not impact the actual generation in any way - in addition, you'll see that the `require_signature_addr` plusarg does in fact only appear once.
Like @rswarbrick suggested, could you please ensure you do not have any local changes and are synced to the top-of-tree when you run?

gxflying, 2020-05-12:
yes ， it's really interesting. I Just had some more tests, and find something more.
I use the latest version of master branch with some modifications. of mine
@udinator.
1. **the "appears twice" problem** is because of my modification. I add the "sim_opts" into the sim command line. and why this modification ? explains below:
from the file configuration.rst (vendor\google_riscv-dv\docs\source line 93) it says that the "sim_opts" is for instruction generator which conflicts with what you said "The sim_opts node you mentioned is actually unnecessary for instruction generation"
add acctually the google dv code does NOT use this option when generates instructions (in do_simulation), **but I think it's necessary**.  while simulation options can be added into gen_opts node, but it's kinds of confusing!  or be added from Makefile(the situation now), it may be not flexible enough(think about the senario tests with different sim opts , say overrided sequences which is my case ).
[image]
2. **about the sim_opts for rtl_sim** 
    as said above, the sim opts of rtl_sim may use other names than "sim_opts" in long term.
     the "google dv" + "ibex" runs good for now. it may not always share the same sim opts between 
   "rtl_sim"  & "gen"(do_simulation in run.py)
3. **about the yaml format**
    @rswarbrick we must keep in mind the difference between the syntax "sim_opts:" and "sim_opts: >"
    and the node "sim_opts" in line 225 does contain one more space line(0x0a) when using 
"sim_opts: >".
    it runs good because in rtl_sim() the code do some more actions by calling replace() , and I forgot it, so error in my case.
   sim_cmd = (base_cmd + ' ' + test['sim_opts'].**replace('\n', ' '**) (line 214 in sim.py)
[image]

in all, the issue itself is clear, it's because of modification that adding sim opts when generating instructions .
it also shows the code may lack of the "sim opts" for "make gen" described above (the first point)

 thanks.

udinator, 2020-05-12:
Ahh, I think I understand the confusion here - basically there are two separate `sim_opts` variables that you are referring to interchangeably.

The first instance of `sim_opts` is, as you point out, in the instruction generator as a command line parameter to the `vendor/google_riscv-dv/run.py` script; this one allows you to pass in instruction generation options.

The second instance of `sim_opts` is in the overall Ibex DV environment as a command line parameter to the `dv/uvm/core_ibex/sim.py` script; this one allows you to pass in RTL simulation-time options.

As you have noticed, we use yaml targets for each test in the Ibex regression suite (the `riscv_dret_test` is but one example) - each yaml target has a node for `gen_opts` and a node for `sim_opts` (keep in mind that each of these yaml targets encompasses the entire flow from instruction generation all the way to RTL simulation).
As a result, we have all options specified in the `gen_opts` node of a yaml target get passed as `sim_opts` to the `vendor/google_riscv-dv/run.py` script of the instruction generator, and we have all options specified in the `sim_opts` node of a yaml target get passed as `sim_opts` to the `dv/uvm/core_ibex/sim.py` script via the Makefile.
To add instruction generation options via the Makefile, you will have to use the `GEN_OPTS` makefile variable to ensure that your other desired options get passed into the instruction generation `run.py` script; similarly if you want to add other RTL simulation-time options via the Makefile, you have to use the `SIM_OPTS` makefile variable.
Hope this clears it up!

Since you've figured out what was going on, I'll close this issue, but feel free to continue this thread if more related issues come up.

closed 2020-05-12 by udinator

gxflying, 2020-05-12:
@udinator thanks for your reply.
as you said,  the key confusing point is all about "sim_opts",
1. I think, it 's may be better to modify the description in configuration.rst (vendor\google_riscv-dv\docs\source line 93). because it says the "sim_opts" in yaml files **NOT** command line is for generating instructions 

Test list in **`YAML format`_**
    .....
    # sim_opts        : Simulation options passed to the instruction generator
2. as I said in the point 1,  sim opts for generation passing in "gen_opts" works, but confusing, because if so, the "gen_opts" acctually contains two things :  opts for instruction feature & opts for simulation behavior.
of course if OK， when we know how the things go on, but this need to look inside the source code.
thanks again!