lowRISC/ibex #857: How to classify and deal with multiple configurations

GregAC, 2020-05-11:
The current multi-config setup for Ibex is simple. We have a number of named configs and they all supply settings for the same parameters, this is one example:

https://github.com/lowRISC/ibex/blob/863fb56eb166dc8d6e25f4e98b7319f0c7088564/ibex_configs.yaml#L10-L16

Are we happy with this setup or do want something more complex? In particular for PMP this isn't really something that pairs nicely with a 'high performance' config you could equally want it with the lower performance or higher performance config. We could define multiple configs one with -pmp one without to have every config (or some subset of them) with pmp on or off (so you'd have a `small-3cmult-pmp-off` and a `small-3cmult-pmp-on`. Alternatively we could introduce config 'add-ons'. Extra bits in the YAML file that could be paired with any config:

```yaml
pmp16-on:
  PMPRegions: 16
  PMPGranularity: 0
```

This would be given to `ibex_config.py` along with the main config name

```
./util/ibex_config.py experimental-maxperf-bm-1cmult pmp16-on fusesoc_opts
```

I was tending towards the config add-ons option, as doing multiple configs with names like '-pmp-on' doesn't scale very well.

(Lets try and divorce this from the what is a supported configuration discussion, my view is we can define supported configs in the documentation and flag them specifically in the `ibex_configs.yaml` but it can still give functionality/parameters for non-supported configs, we could even add a warning output from `ibex_config.py` if you use an unsupported config setup).

rswarbrick, 2020-05-11:
I like the config add-ons approach. It would probably be good if we could also "name" some collections of add-ons, so there would be a "for-opentitan" config etc.

GregAC, 2020-05-11:
Perhaps I should lean more heavily on the add-ons approach and everything becomes an 'add-on' then a 'config' becomes a collection of add-ons and individual parameter settings.

```yaml
high-perf:
  RV32E                    : 0
  RV32M                    : 1
  BranchTargetALU          : 1
  WritebackStage           : 1
  MultiplierImplementation : "single-cycle"

pmp16-on:
  PMPRegions: 16
  PMPGranularity: 0

...

opentitan:
  high-perf
  pmp16-on
  RV32B : 1
```

Or just drop the whole seperation between 'add-ons' and 'configs', everything is a config and configs can include other configs?

rswarbrick, 2020-05-11:
Maybe the "allow inclusions" approach is the best. And allow a list of configurations on the command line, where `./util/ibex_config.py A B C` is equivalent to `./util/ibex_config.py fake_cfg` with

``` yaml
fake_cfg:
  A
  B
  C
```

tomroberts-lowrisc, 2020-05-11:
Configs that can include configs sounds good to me

GregAC, 2020-05-11:
@imphil @vogelpi any further input?

vogelpi, 2020-05-12:
I would try to make this as simple as possible without duplicating stuff. Thus, I don't like the idea of having multiple configurations like `small-3cmult-pmp-off` and `small-3cmult-pmp-on`. If we are going to add a new parameter, we might need to double the configs and things will get messy over time.

Having configs including other configs (not add-ons) seems the way to go.

imphil, 2020-05-12:
I wouldn't to beyond what we have today: a list of configurations.

The config tooling was motivated by the reduction of the test matrix to a simple list. As soon as you add dimensions in the form of includes, sub-configs, etc., you loose that benefit.

I know that this will require us to think hard about configurations. Maybe the thinking could be driven by the question "What do we see as an excellent operating point for Ibex?" and not so much by "What could a user potentially want?" (to which the answer is generally "everything we provide parameters for, and more").

For users who want to go beyond our list of configurations and pick and choose individual bits and pieces we have all SystemVerilog parameters around, together with all testing infrastructure to validate such changes. We can and should give our users as much documentation to make an informed decision about which configurations might work, and which ones are clearly not working. `assert`s in the code can strengthen that message.

GregAC, 2020-05-12:
From our discussion we will be leaving the config setup as is. A straight-forward list of named configs along with the parameter settings for each.

As part of enabling the multi-config DV I'll turn PMP on in at least one of the configs. We'll need to come up with some names as I think adding features with `-feature` to the config name starts to get unwieldy, e.g. we'll have `experimental-maxperf-bm-1cmult-pmpon` if I continue with the current naming scheme when I turn on pmp in the `experimental-maxperf-bm-1cmult` config.

closed 2020-05-12 by GregAC