lowRISC/ibex #859: [dv/makefile] Same seed used for each iteration of Ibex regressions

udinator, 2020-05-11:
Hi,
I've been seeing an issue with Ibex regressions where if I run a test (say riscv_debug_basic_test) 10 times, each of the ten iterations of this test all use the exact same seed since it is now generated in the top level makefile and passed down to every simulation iteration.
This is totally fine for sanity regressions where each test is run exactly once, but for large regressions (nightly, etc...), having every iteration of a test use the exact same seed runs counter to the point of having large numbers of tests in regressions to begin with.
Ideally, if I specify `ITERATIONS=10` on the command line, I expect that each of the ten iterations is using different seeds - I'm not quite sure if having the Makefile generate a top level seed used for every iteration would be the best approach here.
Could you please take a look?
Thanks!

rswarbrick, 2020-05-12:
Hmm. You're right this doesn't quite make sense. I've pushed a fix at instruction generation time here: https://github.com/google/riscv-dv/pull/570. The sim-time fix is in PR #861.

rswarbrick, 2020-05-15:
Tao wasn't keen on defining the combination of `--seed` and `--iterations` > 0 in riscv-dv, so I've amended that patch to use a new flag, `--seed_start`. This will work for us, but needs a minor tweak to how we call riscv-dv when the new code is vendored in. Assuming that https://github.com/google/riscv-dv/pull/570 gets merged in something like it's present form, we'll want a patch like the attached (renamed to .txt because GitHub doesn't allow files called .patch, sigh).

[0001-Pass-start_seed-rather-than-seed-to-riscv-dv-s-run.p.txt](https://github.com/lowRISC/ibex/files/4633553/0001-Pass-start_seed-rather-than-seed-to-riscv-dv-s-run.p.txt)

rswarbrick, 2020-05-15:
Oh, and looking again at what you originally wrote, @udinator, my idea is that we'll pick a starting seed (either at random, or given by the SEED argument to Make). We then run tests with seeds `{SEED, SEED+1, ..., SEED+num_iters}`. Assuming the PRNG isn't completely terrible, this has the same "randomness" as running `num_iters` arbitrary tests, but it lets us make sure the seeds we use in riscv-dv match the seeds we use in the simulator.

Modulo the problem with tests within a batch (see note at bottom of the riscv-dv PR), I think the PRs I've pushed should get everything working for the "Generate 1000 tests with random seeds" nightly use-case.

udinator, 2020-05-15:
SGTM. Could you also amend the Ibex PR to use a `--start_seed` flag as well? The `--seed` flag in sim.py has the same motivation as it's counterpart in run.py, so it would be good to reflect the new method in both scripts. 

Re Makefile changes: you're correct in that we'll just have to change the name of the flag used for generation and simulation from `--seed` to `--start_seed`.

rswarbrick, 2020-05-15:
Ok (I still don't think adding another flag is a good idea, but I agree that we may as well be consistent). I'll wait until the riscv-dv PR has been merged (to make sure Tao is ok with the argument name), because then I can update #861 and vendor in the updated riscv-dv at the same time, which avoids a complicated patch in uvm/core_ibex/Makefile.

closed 2020-05-19 by rswarbrick