lowRISC/ibex #866: Deal with new `sext.b/h` Zbb instructions

vogelpi, 2020-05-13:
As mentioned in #863 , we recently vendored in a riscv-dv version from upstream that features two new instructions added in the [latest working draft version of the bitmanip extension](https://github.com/riscv/riscv-bitmanip/blob/master/bitmanip-draft.pdf) (no version number yet).

These two instructions are not part of the version 0.92 draft implemented in Ibex. As a consequence, the Zbb tests started to fail on master. To work around this, we can either:
1. Implement the new instructions, i.e., go beyond the version 0.92 draft. We might need to do this at some point anyway.
2. Patch the vendored in RISCV-DV files to not emit those instructions, i.e., stay at the version 0.92 draft.

I would prefer Option 2 at least until the latest working draft gets a version number.

ganoam, 2020-05-13:
The spec version 0.92 does include the sext instructions. They are even explicitly mentioned in the change history. But they are not categorized in the table on page 3. I think this is an oversight, and they should really have been part of zbb already. 

I think, if riscv-dv chose to include it already, it would make more sense for us to do it too. They are pretty simple to implement.

Sidenote: it's not the Zbb tests that fail, but Zbe, as RISCV-DV have categorized them in Zbe for now. However they should probably be moved to Zbb as in the latest working draft.

vogelpi, 2020-05-13:
Ah, I see thanks for clarifying that. In this case, it indeed makes sense to implement them at some point.

I will get in touch with DV folks to move them to Zbb. I don't see why they should be part of Zbe.

vogelpi, 2020-05-14:
Update: this will be solved upstream by creating a new Zb_tmp instruction group for the two instructions, see google/riscv-dv#573.

ganoam, 2020-05-22:
This has been merged in #870

closed 2020-05-22 by imphil