lowRISC/ibex #867: [synth] Feedthrough paths on data bus interface

tomroberts-lowrisc, 2020-05-13:
With the Writeback stage enabled, there is a feedthrough path from ``data_rvalid_i`` and ``data_err_i`` back to ``data_req_o`` 

``lsu_resp_valid_o`` --> ``outstanding_memory_access`` --> ``data_req_allowed`` --> ``lsu_req``
``data_err_i`` --> ``wb_exception`` --> ``instr_kill`` --> ``instr_executing`` --> ``lsu_req``

GregAC, 2020-05-13:
This may be a tricky one. If we've got back to back loads then the second load shouldn't execute (i.e. send its request) until it knows the first one has succeeded, it can't do this until it knows the response to the first load (to see if it was an error response and hence whether or not the first load will take an exception). We could always prevent the second load from sending its request until the first load has actually retired, thus cutting the path, but when we have a single cycle memory response we'd like to be able to execute back to back loads. If we make a load wait for a previous one to retire first we'll get stall cycles.

We could allow the second load to just start anyway, at which point Ibex will be generate two outstanding accesses at once. It can already do this but currently limited to misaligned cases where both requests are associated to one instruction, this will cause two unrelated loads/instructions to be outstanding at once so this would be a new behavior. This would be something we'd want to control with a PMA setup (so allow the speculative load to normal memory but not to device memory).

There's also the question of stores which have the same issue, however there we can't speculatively start a second store until the first has completed. Perhaps as with the load we could allow the second speculative store at which point a store failure becomes an imprecise exception (which have been pondering anyway).

For performance I think we either need to leave this path or, or introduce the speculative behaviors described above.

GregAC, 2020-05-13:
I could also try a quick RTL hack to stall between back to back memory accesses and run coremark/embench and see what the impact of that is.

GregAC, 2020-05-19:
The performance impact of stalling to stop back to back memory accesses is significant:

* Coremark without stall: 3.09, with stall: 2.92, 5.8% change
* 8.3% slowdown in Embench

So I don't think the simple fix is a good idea, I do note the cv32e40p (previously RI5CY) contains a very similar path:

https://github.com/openhwgroup/cv32e40p/blob/dde01442d11123763dccdf030b4bd3fbf4d9b4fe/rtl/riscv_load_store_unit.sv#L387-L421

Simply leaving it in could be a possibility, we should do some further timing investigation to determine how much of a potential issue it is, was this causing an issue in OpenTitan FPGA timing?

tomroberts-lowrisc, 2020-05-19:
Agreed that seems like too much of a performance hit.
It's not an issue in the FPGA at the moment. It starts to show up if we remove the pipelining on the d-side, but I think it's still meeting the target frequency.

GregAC, 2020-05-19:
Ultimately we do need to remove pipelining, for the SRAM data access at least, to ensure good performance (i.e. data/response return cycle after request). I think we may end up with a split setup where the data SRAM is like a tightly coupled memory with as little logic between it and the Ibex IO as possible with everything else accessible via the memory system sitting behind some pipeline logic.

GregAC, 2020-05-19:
I've taken a look with a yosys run, this path is 9 levels of logic, which doesn't seem too bad. I think we're going to have to allow some feedthroughs in Ibex, we should come up some guidance around how long they can be and ensuring they are documented.

rswarbrick, 2021-01-27:
So we can make some progress with this, would it make sense to document the existing feedthrough paths? (And maybe even run something in CI to check we don't gain any).

@GregAC, @tomroberts-lowrisc: What do you think?

rswarbrick, 2021-01-27:
The "maybe even run something in CI" part is tracked in #651.

closed 2021-12-10 by GregAC