lowRISC/ibex #871: How to use (openocd + verilator + gdb) to do debug simulation for ibex?

hyf6661669, 2020-05-15:
In https://github.com/PrincetonUniversity/openpiton, they only introduce how to use openpiton with ariane. Is there any guidance? Thanks!

GregAC, 2020-05-15:
We don't provide debug infrastructure in the Ibex repository, however the OpenTitan project that uses Ibex includes debug infrastructure https://github.com/lowRISC/opentitan

We use the debug module from the PULP platform https://github.com/pulp-platform/riscv-dbg

You may also be interested in this PR (which ultimately didn't get merged) https://github.com/lowRISC/ibex/pull/654 which extracts just the bits of OpenTitan required to get the debug infrastructure running.

As for OpenPiton we don't have any support or example of using Ibex within that platform.

GregAC, 2020-05-15:
There is also some brief info on what's required for debug support in Ibex in the documentation here: https://ibex-core.readthedocs.io/en/latest/debug.html

closed 2020-05-17 by hyf6661669