lowRISC/ibex #879: [RTL] Overlapping case statements

sriyerg, 2020-05-18:
Ran into this warning in OpenTitan chip level sim:

```
Warning-[RT-MTOCMUCS] More than one condition match in statement
../src/lowrisc_ibex_ibex_core_0.1/rtl/ibex_alu.sv, 78
  More than one condition matches are found in 'unique case' statement inside 
  tb.dut.top_earlgrey.u_rv_core_ibex.u_core.ex_block_i.alu_i, at time 
  234570001 psps.
  
  Line number 79 and 80 are overlapping
```

Steps to reproduce:
- Clone OpenTitan repo, cd to repo top
- Checkout PR 2052 if it is not already merged
- Run chip DV sim:
```
$ util/dvsim/dvsim.py -i hw/top_earlgrey/dv/chip_sim_cfg.hjson -i chip_uart_tx_rx [--waves] [--build-modes en_ibex_tracer]
```

PTAL, thanks!

rswarbrick, 2020-05-19:
I think this is probably a duplicate of #845, which was fixed in the Ibex repository by commit `4f349a`. Can you try again, passing the `-xlrm uniq_prior_final` arguments to VCS? If that silences the message, you'll need to add the equivalent arguments for OpenTitan.

sriyerg, 2020-05-19:
Okay, thanks!

sriyerg, 2020-05-19:
Duplicate of #845, closing.

closed 2020-05-19 by sriyerg