lowRISC/ibex #880: Beginner question: C.JR not branching

elchanan18, 2020-05-19:
I'm trying to understand why the branch isn't happening.
On cycle 9/10, we excecute a C.LI which loads value 0 into reg 0F.
On cycle 11/12, we execute a C.JR with reg 0F (which contains 0),
so we'd expect rvfi_pc_rdata to be 0 on cycle 13/14, not to just increment to DDFFFF84.

Is there something obvious I'm missing?
Thank you

[image]

tomroberts-lowrisc, 2020-05-19:
Hi there
Tricky to tell what's happening from just the picture. If you dump the CEX trace as a vcd and attach it here I can help you debug it.

elchanan18, 2020-05-19:
Wow.  Thanks for the quick response!
But it won't let me post a .vcd

[image]

tomroberts-lowrisc, 2020-05-19:
you can zip it then attach it

elchanan18, 2020-05-19:
[cjr.zip](https://github.com/lowRISC/ibex/files/4650589/cjr.zip)

tomroberts-lowrisc, 2020-05-19:
Thanks - how are you decoding/determining those opcodes?
I decode these as:
``0x4781 = li x15, 0``
``0x87CE = mv x15, x19``
``0xDFEAD9E3 = bge x21, x30, 0xFFFFFDF2``

elchanan18, 2020-05-19:
Oh.  Reading the ISA for C.JR (page 105) doesn't mention that rs2 has to be x0.  Now, looking at the ISA for C.MV (page 108), I see where it adds that rs2=x0 points to C.JR
My bad.
I should have checked first with the table on page 113.
Thank you for the lesson.

tomroberts-lowrisc, 2020-05-19:
No problem - glad you figured it out
I'll close this issue now, feel free to reopen if you have further questions.

closed 2020-05-19 by tomroberts-lowrisc