lowRISC/ibex #909: PMP: Configurable default PMP configuration

imphil, 2020-05-28:
With the default action in M mode being ALLOW, PMPs can be used only to disable access to regions, not to enable regions step-by-step.

To iteratively enable execution, this extension would set a default policy for M mode, e.g. enable execution only out of ROM after a reset.

The goal of this issue is to fully specify and try to implement this idea. If we figure out it makes sense to do something like that, we should get in touch with the RISC-V standards bodies to get their feedback and potentially include it in the RISC-V spec.

(Another option might be a "PMP 0" which is defined at synthesis time, but appears as a normal PMP entry. This becomes more of an option if we have more than 16 PMP entires available, see #908.)

imphil, 2020-10-23:
Instead of doing something custom, we've decided to go with the ePMP spec. @GregAC, do you agree that this issue isn't relevant any more in that case?

GregAC, 2020-10-23:
Yup ePMP gives us what we need here.

closed 2020-10-23 by GregAC