lowRISC/ibex #910: PMP: Allow the reduction of permissions in locked entries

imphil, 2020-05-28:
This is a potential extension to RISC-V standardized PMPs.

Let's see if we can implement a reduction of permissions (e.g. from RWXL to RL) even for locked PMP entries. Doing that should allow people to re-use previous locked PMP entries in limited ways.

If we come to the conclusion that an implementation is feasible and a good idea, we should get in touch with the RISC-V standardization groups to discuss and ideally include it in the spec.

vogelpi, 2021-04-13:
Related to this issue, support for [ePMP proposed by the RISC-V TEE Task Group](https://ibex-core.readthedocs.io/en/latest/_downloads/adbc55b5b9bad5f92968b58e7b12fc57/riscv-epmp.pdf) got implemented in PR #1232. This enables somewhat similar behavior: With Rule Locking Bypass `mseccfg.RLB` it is temporary possible to edit locked entries, e.g., during the boot process.

However, the original idea of further reducing permissions later on still seems interesting. I don't know if anything around this has been discussed when deciding to go ahead and implement ePMP. It could be there is a problem with that idea. Maybe @GregAC know more? Anyway, I think it makes total sense for Ibex to stick to the ePMP proposal for the time being.

imphil, 2021-04-13:
Since this proposal was written two things changed, which make it not as pressing/relevant any more:
* Support for more than 16 PMPs entries was standardized, reducing the need to re-use entries (at least from a programmer's point of view, there could still be value from a resource utilization point of view).
* ePMP came along, which also covers some functionality this proposal would cover.

Since we have no intention of implementing the behavior described here in the moment, let's close the issue.

closed 2021-04-13 by imphil