lowRISC/ibex #912: Back to back exceptions

udinator, 2020-05-29:
Hi,
I found an interesting issue when running a PMP test, the waves are shown below:
[image]

The situation is as follows.
The instruction at `PC=0x8000921c` raises a PMP exception trying to store data to address `0x80014d04`, which is correctly propagated through the LSU and to the controller, which sees its `store_err_i` and `store_err_q` go high at the yellow marker.
The very next instruction at `PC=80009220` raises a PMP fetch exception in the IF stage, which again gets propagated to the controller correctly, as its `instr_fetch_err_i` and `instr_fetch_err` signals both go high.
The problem here arises due to the fact that `instr_fetch_err` and `store_err_q` are both high in the controller during the cycle when it transitions to the `FLUSH` state to handle an exception, causing the core to re spond to the instruction fetch exception and ignore the store fault completely due to exception prioritization.

I'm pretty certain that this can be recreated with non-PMP exceptions as it just requires exceptions raised by two sequential instructions (as long as one is a memory-related exception) to be seen by the controller on the same cycle.

GregAC, 2020-05-29:
I'll take a look at this

GregAC, 2020-05-29:
This is an issue introduced by the 3rd pipeline stage. Whilst for the same instruction an instruction fault clearly take priority over a store fault (as you cannot execute the instruction to produce a store fault if there's an instruction fault) when you have an instruction in the writeback stage any load/store fault it sees takes priority over the instruction fault seen in the instruction following. The controller fault prioritization doesn't currently understand this so the instruction fault ends up taking priority.

GregAC, 2020-06-01:
Whilst investigating this found we have another bug where if there was an outstanding load or store in writeback that had yet to get a response (so we don't know if it will cause an exception) but an instruction in ID/EX sees something that would cause an exception (e.g. a PMP failure for the instruction or an invalid instruction) we'd immediately trigger the exception for ID/EX.

Instead we need to ensure that when triggering an exception for ID/EX that anything in WB has finished so any exception it raises is correctly dealt with.

#919 fixes both the issue here and the one I've just mentioned

commit 833bdefe (2020-06-02): [rtl] Fix exception priority for writeback stage

With the writeback stage enabled the controller can see a load or store
error from the writeback stage whilst seeing some other fault/exception
from ID/EX (e.g. an instruction fetch error). The writeback stage fault
must take priority, however without the writeback stage the
priortisation changes.

This introduces more explicit prioritisation logic for faults/exceptions
and gives the correct prioritisation for configurations both with and
without a writeback stage.

Fixes #912
closed 2020-06-02 by GregAC