lowRISC/ibex #931: Missing dependencies on lowrisc:prim:assert

flaviens, 2020-06-07:
Hi all,

Apparently, there are some dependencies missing to lowrisc:prim:assert in core configuration files, notably in lowrisc:prim:ram_1p and lowrisc:prim:ram_2p.
This may have remained undetected, since the reference to these cores are always done by lowrisc:prim:all.

All my apologies if I did a mistake or if this is a duplicate.

rswarbrick, 2020-06-07:
I agree that something looks awry, but I'm not quite sure about the diagnosis. Taking the 1-port RAMs to start with, `lowrisc:prim:ram_1p` defines a thin wrapper (in `prim_ram_1p.sv`) around `prim_generic_ram_1p`, which comes from `lowrisc:prim_generic:ram_1p`. It turns out that `prim_generic_ram_1p.sv` *does* include `prim_assert.svh` and `prim_generic_ram_1p.core` doesn't list the assertion (tsk, tsk). But it seems the assertions are never actually used! So I think the correct fix would be to remove the include. I think the same is true about the 2-port version.

Since this code is vendored from OpenTitan, I'm afraid the PR will have to go there.

rswarbrick, 2020-07-03:
Fixed upstream with https://github.com/lowRISC/opentitan/pull/2700 (which will need vendoring back in to Ibex when it's merged).

closed 2020-07-06 by rswarbrick