lowRISC/ibex #954: UVM DV: Unsynchronized DUT probe signals

tomroberts-lowrisc, 2020-06-15:
In PR #953 I corrected some paths that were causing spurious triggering of internal TB events. Even with that fix, the spurious triggers remain due to glitchy sampling of those signals.

I hacked around it by replacing all instances of `wait (signal);` with `while (!signal) @(posedge clk);` but I guess the correct fix is a clocking block.

udinator, 2020-06-15:
Thanks @tomroberts-lowrisc for filing this issue!
You're absolutely right, it's better in general to synchronize interface signal sampling with clocking blocks. I'll work on this when I have some time and get a PR open to add clocking blocks in the Ibex testbench.

udinator, 2020-06-22:
#978 has been merged, which synchronizes all Ibex interfaces with clocking blocks.
@tomroberts-lowrisc do you think you could rerun the failing seeds with the top-of-tree? From what you mentioned, there might be a possibility that adding clocking blocks is only a partial solution to the failures you encountered.

tomroberts-lowrisc, 2020-06-23:
Hi @udinator - I reran at my end. All tests passing except debug_branch_jump and debug_single_step, which I think are due to unrelated issues.

closed 2020-06-23 by tomroberts-lowrisc